AsteroidOS/asteroid #3: Cannot build for dory

TheLastProject, 2016-07-11:
Debian unstable

```
ERROR: oe_runmake failed
ERROR: Function failed: do_configure (log file is located at /home/sylvia/Programs/asteroid/build/tmp-glibc/work/armv7a-vfp-neon-oe-linux-gnueabi/libiodata/+git5c7ddd20667e09c877cbbc48bab57942e6bd9897-r1/temp/log.do_configure.32344)
ERROR: Logfile of failure stored in: /home/sylvia/Programs/asteroid/build/tmp-glibc/work/armv7a-vfp-neon-oe-linux-gnueabi/libiodata/+git5c7ddd20667e09c877cbbc48bab57942e6bd9897-r1/temp/log.do_configure.32344
Log data follows:
| DEBUG: Executing python function sysroot_cleansstate
| DEBUG: Python function sysroot_cleansstate finished
| DEBUG: Executing shell function qmake5_base_preconfigure
| DEBUG: Shell function qmake5_base_preconfigure finished
| DEBUG: Executing shell function do_configure
| Cannot read /home/sylvia/Programs/asteroid/build/tmp-glibc/sysroots/dory/usr/lib/mkspecs/oe-device-extra.pri: No such file or directory
| NOTE: make MAKEFLAGS=-j 2 OE_QMAKE_COMPILER=arm-oe-linux-gnueabi-gcc  -march=armv7-a -mfloat-abi=softfp -mfpu=neon --sysroot=/home/sylvia/Programs/asteroid/build/tmp-glibc/sysroots/dory OE_QMAKE_CC=arm-oe-linux-gnueabi-gcc  -march=armv7-a -mfloat-abi=softfp -mfpu=neon --sysroot=/home/sylvia/Programs/asteroid/build/tmp-glibc/sysroots/dory OE_QMAKE_CXX=arm-oe-linux-gnueabi-g++  -march=armv7-a -mfloat-abi=softfp -mfpu=neon --sysroot=/home/sylvia/Programs/asteroid/build/tmp-glibc/sysroots/dory OE_QMAKE_CFLAGS= -O2 -pipe -g -feliminate-unused-debug-types OE_QMAKE_CXXFLAGS= -O2 -pipe -g -feliminate-unused-debug-types -fvisibility-inlines-hidden OE_QMAKE_LINK=arm-oe-linux-gnueabi-g++  -march=armv7-a -mfloat-abi=softfp -mfpu=neon --sysroot=/home/sylvia/Programs/asteroid/build/tmp-glibc/sysroots/dory OE_QMAKE_LDFLAGS=-Wl,-O1 -Wl,--hash-style=gnu -Wl,--as-needed OE_QMAKE_AR=arm-oe-linux-gnueabi-ar OE_QMAKE_STRIP=echo OE_QMAKE_WAYLAND_SCANNER=/home/sylvia/Programs/asteroid/build/tmp-glibc/sysroots/x86_64-linux/usr/bin/wayland-scanner OE_QMAKE_INCDIR_QT=/home/sylvia/Programs/asteroid/build/tmp-glibc/sysroots/dory//usr/include
| arm-oe-linux-gnueabi-g++  -march=armv7-a -mfloat-abi=softfp -mfpu=neon --sysroot=/home/sylvia/Programs/asteroid/build/tmp-glibc/sysroots/dory -c -pipe -O2 -pipe -g -feliminate-unused-debug-types -fvisibility-inlines-hidden -O2 -std=gnu++0x -Wall -W -D_REENTRANT -fPIC -DQT_NO_DEBUG -DQT_CORE_LIB -I. -isystem /home/sylvia/Programs/asteroid/build/tmp-glibc/sysroots/dory/usr/include/QtCore -I. -I/home/sylvia/Programs/asteroid/build/tmp-glibc/sysroots/dory/usr/lib/mkspecs/linux-oe-g++ -o validator.o validator.cpp
| arm-oe-linux-gnueabi-g++  -march=armv7-a -mfloat-abi=softfp -mfpu=neon --sysroot=/home/sylvia/Programs/asteroid/build/tmp-glibc/sysroots/dory -c -pipe -O2 -pipe -g -feliminate-unused-debug-types -fvisibility-inlines-hidden -O2 -std=gnu++0x -Wall -W -D_REENTRANT -fPIC -DQT_NO_DEBUG -DQT_CORE_LIB -I. -isystem /home/sylvia/Programs/asteroid/build/tmp-glibc/sysroots/dory/usr/include/QtCore -I. -I/home/sylvia/Programs/asteroid/build/tmp-glibc/sysroots/dory/usr/lib/mkspecs/linux-oe-g++ -o storage.o storage.cpp
| In file included from storage.cpp:33:0:
| storage.cpp: In member function 'void iodata::storage::set_primary_path(const string&)':
| log.h:88:29: error: unable to find string literal operator 'operator""ARGS' with 'const char [9]', 'unsigned int' arguments
|              fprintf(stderr, "ASSERT: "ARGS); \
|                              ^
| storage.cpp:55:3: note: in expansion of macro 'log_assert'
|    log_assert(path.size()==0, "primary path '%s' already defined", path[0].c_str()) ;
|    ^
| storage.cpp: In member function 'void iodata::storage::set_secondary_path(const string&)':
| log.h:88:29: error: unable to find string literal operator 'operator""ARGS' with 'const char [9]', 'unsigned int' arguments
|              fprintf(stderr, "ASSERT: "ARGS); \
|                              ^
| storage.cpp:61:3: note: in expansion of macro 'log_assert'
|    log_assert(path.size()<2, "secondary path '%s' already defined", path[1].c_str()) ;
|    ^
| log.h:88:29: error: unable to find string literal operator 'operator""ARGS' with 'const char [9]', 'unsigned int' arguments
|              fprintf(stderr, "ASSERT: "ARGS); \
|                              ^
| storage.cpp:62:3: note: in expansion of macro 'log_assert'
|    log_assert(path.size()>0, "can't set secondary path, because primary path not set") ;
|    ^
| log.h:88:29: error: unable to find string literal operator 'operator""ARGS' with 'const char [9]', 'unsigned int' arguments
|              fprintf(stderr, "ASSERT: "ARGS); \
|                              ^
| storage.cpp:63:3: note: in expansion of macro 'log_assert'
|    log_assert(path.size()==1) ;
|    ^
| storage.cpp: In member function 'void iodata::storage::set_validator(const string&, const string&)':
| log.h:88:29: error: unable to find string literal operator 'operator""ARGS' with 'const char [9]', 'unsigned int' arguments
|              fprintf(stderr, "ASSERT: "ARGS); \
|                              ^
| storage.cpp:69:3: note: in expansion of macro 'log_assert'
|    log_assert(!name.empty()) ;
|    ^
| log.h:88:29: error: unable to find string literal operator 'operator""ARGS' with 'const char [9]', 'unsigned int' arguments
|              fprintf(stderr, "ASSERT: "ARGS); \
|                              ^
| storage.cpp:70:3: note: in expansion of macro 'log_assert'
|    log_assert(type_validator==NULL) ;
|    ^
| storage.cpp: In member function 'void iodata::storage::set_validator(iodata::validator*, const string&)':
| log.h:88:29: error: unable to find string literal operator 'operator""ARGS' with 'const char [9]', 'unsigned int' arguments
|              fprintf(stderr, "ASSERT: "ARGS); \
|                              ^
| storage.cpp:78:3: note: in expansion of macro 'log_assert'
|    log_assert(!name.empty()) ;
|    ^
| log.h:88:29: error: unable to find string literal operator 'operator""ARGS' with 'const char [9]', 'unsigned int' arguments
|              fprintf(stderr, "ASSERT: "ARGS); \
|                              ^
| storage.cpp:79:3: note: in expansion of macro 'log_assert'
|    log_assert(type_validator==NULL) ;
|    ^
| storage.cpp: In member function 'iodata::record* iodata::storage::load()':
[... 98 lines cut ...]
| storage.cpp: In static member function 'static int iodata::storage::read_file_to_string(const char*, std::__cxx11::string&)':
| log.h:88:29: error: unable to find string literal operator 'operator""ARGS' with 'const char [9]', 'unsigned int' arguments
|              fprintf(stderr, "ASSERT: "ARGS); \
|                              ^
| storage.cpp:407:3: note: in expansion of macro 'log_assert'
|    log_assert(buffer) ;
|    ^
| In file included from validator.cpp:28:0:
| validator.cpp: In member function 'void iodata::validator::load(const iodata::record*)':
| log.h:88:29: error: unable to find string literal operator 'operator""ARGS' with 'const char [9]', 'unsigned int' arguments
|              fprintf(stderr, "ASSERT: "ARGS); \
|                              ^
| validator.cpp:312:7: note: in expansion of macro 'log_assert'
|        log_assert(n) ;
|        ^
| Makefile:392: recipe for target 'storage.o' failed
| make: *** [storage.o] Error 1
| make: *** Waiting for unfinished jobs....
| Makefile:386: recipe for target 'validator.o' failed
| make: *** [validator.o] Error 1
| WARNING: exit code 1 from a shell command.
| ERROR: oe_runmake failed
| ERROR: Function failed: do_configure (log file is located at /home/sylvia/Programs/asteroid/build/tmp-glibc/work/armv7a-vfp-neon-oe-linux-gnueabi/libiodata/+git5c7ddd20667e09c877cbbc48bab57942e6bd9897-r1/temp/log.do_configure.32344)
ERROR: Task 1936 (/home/sylvia/Programs/asteroid/src/meta-asteroid/recipes-nemomobile/libiodata/libiodata_git.bb, do_configure) failed with exit code '1'
```

Lrs121, 2016-07-11:
perform a git pull on src/meta-asteroid and see if that helps. the recipe for libiodata was modified

TheLastProject, 2016-07-12:
That seems to have fixed the build. However, I am unable to push the new version to the device:
```
$ adb push -p build/tmp-glibc/deploy/images/dory/asteroid-image-dory.ext2 /sdcard/linux/rootfs.ext2                       
* daemon not running. starting it now on port 5037 *
* daemon started successfully *
Transferring: 355574784/355574784 (100%)
failed to copy 'build/tmp-glibc/deploy/images/dory/asteroid-image-dory.ext2' to '/sdcard/linux/rootfs.ext2': No space left on device
```

```
$ ls -lah build/tmp-glibc/deploy/images/dory/asteroid-image-dory.ext2                                                          
lrwxrwxrwx 1 sylvia sylvia 46 Jul 12 06:40 build/tmp-glibc/deploy/images/dory/asteroid-image-dory.ext2 -> asteroid-image-dory-20160712040600.rootfs.ext2
$ ls -lah build/tmp-glibc/deploy/images/dory/asteroid-image-dory-20160712040600.rootfs.ext2 
-rw-r--r-- 1 sylvia sylvia 340M Jul 12 06:40 build/tmp-glibc/deploy/images/dory/asteroid-image-dory-20160712040600.rootfs.ext2
```

Lrs121, 2016-07-12:
This line right there tells you the problem. "failed to copy 'build/tmp-glibc/deploy/images/dory/asteroid-image-dory.ext2' to '/sdcard/linux/rootfs.ext2': No space left on device"

More specifically it's saying you don't have enough available storage space on the watch. From your ls you can tell that you need approximately 340 MB free. I'd make that ~400 MB to be safe.

TheLastProject, 2016-07-12:
Well, that's the problem. I have never used that device except flash AsteroidOS on it before once. Unless the old AsteroidOS installation lingers around in some way I'm not aware of, there simply cannot be a lack of space on the SD card unless the image has grown too much since the January 22nd build on the website.

FlorentRevest, 2016-07-12:
Fixed by disabling logs in libiodata from this commit https://github.com/AsteroidOS/meta-asteroid/commit/6b9ccc49cb5b69b49db6541244200ec322a0eef8

The push error (discussed on IRC)  was because you were trying to push the rootfs while running asteroid and can't push a rootfs inside "itself". 

Have a nice day !

closed 2016-07-12 by FlorentRevest