AsteroidOS/asteroid #5: Smartwatch 3 build Issues

return, 2016-08-08:
The tetra build for AsteroidOS keeps throwing this error: 

ERROR: Task do_compile in /<path to asteroid folder>/asteroid/src/meta-tetra-hybris/recipes-kernel/linux/linux-tetra_lp-mr1.bb depends upon non-existent task do_image_complete in /<path to asteroid folder>/asteroid/src/meta-smartphone/meta-android/recipes-core/images/initramfs-android-image.bb

I know that Asteroid OS has early support for Smartwatch 3, but am I missing something required for this to build correctly? Here is my log:

NOTE: Resolving any missing task queue dependencies

Build Configuration:
BB_VERSION        = "1.28.0"
BUILD_SYS         = "x86_64-linux"
NATIVELSBSTRING   = "Debian-8.5"
TARGET_SYS        = "arm-oe-linux-gnueabi"
MACHINE           = "tetra"
DISTRO            = "asteroid"
DISTRO_VERSION    = "20160808-snapshot"
TUNE_FEATURES     = "arm armv7a vfp neon"
TARGET_FPU        = "vfp-neon"
meta-qt5          = "krogoth:460df4154eaa2e62ae077f28e5f8433e3b813f83"
meta              = "jethro:a27b907dd3ad20fc60b7732c19012793aaaba2df"
meta-asteroid     = "master:f87711ad01c46117c194f0894a8bd2572bc7f2bf"
meta-oe           
meta-ruby         
meta-xfce         
meta-gnome        = "jethro:8ab04afbffb4bc5184cfe0655049de6f44269990"
meta-android      = "krogoth:333dc591f9493c6eff3fb60f67d01290c9ca7e2b"
meta-python       
meta-filesystems  = "jethro:8ab04afbffb4bc5184cfe0655049de6f44269990"
meta-tetra-hybris = "master:20b86de2bf4e3c90ecae19aa523f4601c4171944"

NOTE: Preparing RunQueue
ERROR: Task do_compile in /<path to asteroid folder>/asteroid/src/meta-tetra-hybris/recipes-kernel/linux/linux-tetra_lp-mr1.bb depends upon non-existent task do_image_complete in /<path to asteroid folder>/asteroid/src/meta-smartphone/meta-android/recipes-core/images/initramfs-android-image.bb

FlorentRevest, 2016-08-08:
Hi return,
This is indeed a known issue, it happens since we use the krogoth branch of meta-smartphone and the jethro branch of oe-core. And this commit made those two branches incompatible https://github.com/shr-distribution/meta-smartphone/commit/3d4cfe5fd389027d8f4564c2e536bfed4fbec156
As a temporary workaround, you can revert it in src/meta-smartphone and it will compile.
The real solution would be to upgrade oe-core to krogoth but the last time I tried it broke plenty of things. Another solution would be to downgrade meta-smartphone to jethro since it doesn't offer anything critically important.

TheLastProject, 2016-09-27:
I think this issue could perhaps be closed since 3b429f8acd031eab30fa32e8184f9ceeee0668ee, unless it's still an issue?

FlorentRevest, 2016-09-27:
Indeed, it was also fixed before thanks to https://github.com/AsteroidOS/meta-asteroid/commit/5dc1eac2d5eab1e7c5fbdeb23071748d765987f8 but now the problem doesn't exist anymore.

closed 2016-09-27 by FlorentRevest