AsteroidOS/asteroid #6: Compilation problems while bitbaking for LG Watch Urbane

kecsap, 2016-10-20:
Build environment: Ubuntu 16.04, 64 bit.

Openssl build fails. Recipe needs -fPIC to the CFLAGS
libaccounts-glib build fails. libaccounts-glib_git.bb needs --disable-man flag to the EXTRA_OECONF

kecsap, 2016-10-20:
Multiple qt components fails to build or install.

This is broken.

FlorentRevest, 2016-10-20:
Hi kecsap,
Thanks for reporting those issues! Regarding OpenSSL, I don't think that meta-asteroid should be the proper place to fix the CFLAGS, I assume that this should/will be fixed upstream by OpenEmbedded.
Regarding libaccounts-glib you are completely right, I'll fix that.
For the Qt components do you have some logs? Once again meta-qt5 might be a better place to report those issues but I haven't often met problems with those recipes so I'm surprised.

Thanks again for the bugreports, it's much appreciated!

kecsap, 2016-10-21:
I have to comment out some lines to get qtbase compile:

kecsap, 2016-10-21:
Something is really wrong here.

Some qt packages (e.g qmake5_base_preconfigure) fail with:

"DEBUG: Executing shell function do_configure
Could not find qmake configuration file linux-oe-g++."

Some (e.g meta-asteroid/recipes-nemomobile/qtaround/qtaround_git.bb) starts to compile, but seems that qt base packages have wrong binary architecture:

/home/kecsap/asteroid/build/tmp-glibc/sysroots/x86_64-linux/usr/bin/arm-oe-linux-gnueabi/arm-oe-linux-gnueabi-g++  -fPIC  -march=armv7ve -mfpu=neon  -mfloat-abi=softfp  --sysroot=/home/kecsap/asteroid/build/tmp-glibc/sysroots/bass  -O2 -pipe -g -feliminate-unused-debug-types -fdebug-prefix-map=/home/kecsap/asteroid/build/tmp-glibc/work/armv7ve-neon-oe-linux-gnueabi/qtaround/+gitbf7f438ea7281d3341f6b353a427a15b2cf10a7c-r1=/usr/src/debug/qtaround/+gitbf7f438ea7281d3341f6b353a427a15b2cf10a7c-r1 -fdebug-prefix-map=/home/kecsap/asteroid/build/tmp-glibc/sysroots/x86_64-linux= -fdebug-prefix-map=/home/kecsap/asteroid/build/tmp-glibc/sysroots/bass=  -fvisibility-inlines-hidden -Wall -Wextra -feliminate-unused-debug-types -Wl,--as-needed -Wl,--no-undefined -Wl,-z,now -std=gnu++11 -Wno-psabi  -Wl,-O1 -Wl,--hash-style=gnu -Wl,--as-needed -shared -Wl,-soname,libqtaround.so.1 -o libqtaround.so.0.2.6 CMakeFiles/qtaround.dir/__/include/qtaround/moc_mt.cpp.o CMakeFiles/qtaround.dir/debug.cpp.o CMakeFiles/qtaround.dir/os.cpp.o CMakeFiles/qtaround.dir/json.cpp.o CMakeFiles/qtaround.dir/sys.cpp.o CMakeFiles/qtaround.dir/subprocess.cpp.o CMakeFiles/qtaround.dir/util.cpp.o CMakeFiles/qtaround.dir/mt.cpp.o CMakeFiles/qtaround.dir/qtaround_automoc.cpp.o /home/kecsap/asteroid/build/tmp-glibc/sysroots/x86_64-linux/usr/lib/libQt5Core.so.5.6.2 -lcor -lcor -Wl,-rpath,/home/kecsap/asteroid/build/tmp-glibc/sysroots/x86_64-linux/usr/lib:
| /home/kecsap/asteroid/build/tmp-glibc/sysroots/x86_64-linux/usr/lib/libQt5DBus.so.5.6.2: file not recognized: File format not recognized

FlorentRevest, 2016-10-21:
Have you stopped the compilation abruptly at some point?
You can at the very least try to "bitbake -c cleanall qtbase qtbase-native" to regenerate some of the incorrect files

I don't think libQt5DBus has been compiled for the wrong architecture, bitbake checks this kind of things in the QA step and it would have yelled before. Anyway if you want to make sure you can use file to check the architecture of the binary.

theappleman, 2016-10-21:
Just to put some context into that, here is the build configuration I had for my tests:

```
~/asteroid/build$ bitbake asteroid-image
Build Configuration:
BB_VERSION        = "1.30.0"
BUILD_SYS         = "x86_64-linux"
NATIVELSBSTRING   = "Ubuntu-16.04"
TARGET_SYS        = "arm-oe-linux-gnueabi"
MACHINE           = "bass"
DISTRO            = "asteroid"
DISTRO_VERSION    = "20161020-snapshot"
TUNE_FEATURES     = "arm armv7ve vfp  neon"
TARGET_FPU        = "softfp"
meta-qt5          = "krogoth:1ec776771f5f09c97917d3839d8140f9a7668c4a"
meta              = "krogoth:a8f44dff13481feaa97e494a3aeafb5b63d40f3f"
meta-asteroid     = "master:bacfe100a488f42240c7770a1660838a06024651"
meta-oe           
meta-ruby         
meta-xfce         
meta-gnome        = "krogoth:851a064b53dca3b14dd33eaaaca9573b1a36bf0e"
meta-android      = "krogoth:e1700cf3ae5226262c9ac348c225251f8a25a103"
meta-python       
meta-filesystems  = "krogoth:851a064b53dca3b14dd33eaaaca9573b1a36bf0e"
meta-sparrow-hybris = "master:51bd5b65874a98c871b8bd65502dc69fbf1dad78"
meta-tetra-hybris = "master:fc87999d68350da334ca2ab399120d0d53d87c7c"
meta-bass-hybris  = "master:21961d51a654052b22e95f8664c8c5161f3d1373"
meta-dory-hybris  = "master:6adc36750417e5239512915c5771b322c78466bc"

NOTE: Preparing RunQueue
NOTE: Executing SetScene Tasks
NOTE: Executing RunQueue Tasks
WARNING: brcm-patchram-plus-4.1.1+gitrAUTOINC+69371f6ecc-r0 do_fetch: Failed to fetch URL git://git.freesmartphone.org/utilities.git;protocol=git;branch=master, attempting MIRRORS if available
NOTE: Tasks Summary: Attempted 3626 tasks of which 2457 didn't need to be rerun and all succeeded.

Summary: There was 1 WARNING message shown.
```

kecsap, 2016-10-22:
Hi,

now everything went well after I started everything from scratch. One list thing, the bass install instructions has "dory" in the path names, but it should be "bass" for flashing and fastboot.

FlorentRevest, 2016-10-22:
Good catch, it's fixed, thank you!

kecsap, 2016-10-23:
Ok, closing.

closed 2016-10-23 by kecsap