AsteroidOS/asteroid #9: Build failure due to missing mirror: git.freesmartphone.org

bencord0, 2016-12-28:
Part of the asteroid build system requires fetching code from `git://git.freesmartphone.org`. However, I am getting the following failure.

```
ERROR: brcm-patchram-plus-4.1.1+gitrAUTOINC+69371f6ecc-r0 do_fetch: Fetcher failure for URL: 'git://git.freesmartphone.org/utilities.git;protocol=git;branch=master'. Unable to fetch URL from any source.
```

## Steps to reproduce

```
bitbake -c fetch brcm-patchram-plus
```

## Expected results
The code is downloaded and saved locally

## Actual results
See above error, DNS fails for me.

```
Cloning into bare repository '[blah blah]/asteroid/asteroid/build/downloads/git2/git.freesmartphone.org.utilities.git'...
fatal: Unable to look up git.freesmartphone.org (port 9418) (No address associated with hostname)
```

bencord0, 2016-12-28:
If someone has a local checkout, we may need to mirror this repo under the github.com/AsteroidOS org.

FlorentRevest, 2016-12-28:
Try with https://github.com/AsteroidOS/meta-asteroid/commit/2e9671741a3618e097657c80761c63945558c701

bencord0, 2016-12-28:
Fixed as of https://github.com/AsteroidOS/meta-asteroid/commit/d62eff58ee59287f50f266ed16745454b21d71e9

closed 2016-12-28 by bencord0