AsteroidOS/asteroid #10: Build sdk fails

aebruno, 2017-01-09:
I was able to successfully bitbake the asteroid-image (bass) and run on the LG Urbane.

I'm now trying to build the sdk but it fails with this error. I know the docs say this is broken but just thought I'd post and see if there's any pointers to get this working. I'm interested in developing some apps. Thanks in advance.

Here's the error:

```
$ . ./prepare-build.sh bass
$ bitbake meta-toolchain-qt5
Loading cache: 100% |#####################################################################################################################################################| Time: 0:00:00
Loaded 2699 entries from dependency cache.
Parsing recipes: 100% |###################################################################################################################################################| Time: 0:00:01
Parsing of 2009 .bb files complete (1997 cached, 12 parsed). 2711 targets, 445 skipped, 0 masked, 0 errors.
NOTE: Resolving any missing task queue dependencies

Build Configuration:
BB_VERSION        = "1.32.0"
BUILD_SYS         = "x86_64-linux"
NATIVELSBSTRING   = "Debian-8.6"
TARGET_SYS        = "arm-oe-linux-gnueabi"
MACHINE           = "bass"
DISTRO            = "asteroid"
DISTRO_VERSION    = "alpha-1.0"
TUNE_FEATURES     = "arm armv7ve vfp  neon"
TARGET_FPU        = "softfp"
meta-qt5          = "krogoth:2b1871f0d139dc3caaa779a32a1931409c245a36"
meta              = "morty:c8d96b10ee3bc2eae0fd269d2564286fd0bc82ed"
meta-asteroid     = "master:d62eff58ee59287f50f266ed16745454b21d71e9"
meta-oe
meta-ruby
meta-xfce
meta-gnome        = "morty:1efa5d623bc64659b57389e50be2568b1355d5f7"
meta-android      = "morty:5476172128e1592e13e4ed7c3f8af785054bec0f"
meta-python
meta-filesystems  = "morty:1efa5d623bc64659b57389e50be2568b1355d5f7"
meta-sparrow-hybris = "master:a98035019439ffbb10a50ba96c5ace573f12096e"
meta-tetra-hybris = "master:e05573fcd910f266df8a8ddab33631fab836b6b5"
meta-bass-hybris  = "master:c52814df8f6f0a131dfb9c6a486ab4ff5205ae69"
meta-dory-hybris  = "master:21bbc30441d99c2cb2f318c29c51f4b043ab198d"

Initialising tasks: 100% |################################################################################################################################################| Time: 0:00:06
NOTE: Executing SetScene Tasks
NOTE: Executing RunQueue Tasks
ERROR: qtwebkit-5.6.2+gitAUTOINC+bb43bca166-r0 do_compile: oe_runmake failed
ERROR: qtwebkit-5.6.2+gitAUTOINC+bb43bca166-r0 do_compile: Function failed: do_compile (log file is located at /tmp/asteroid/build/tmp-glibc/work/armv7ve-neon-oe-linux-gnueabi/qtwebkit/5.6.2+gitAUTOINC+bb43bca166-r0/temp/log.do_compile.17240)
ERROR: Logfile of failure stored in: /tmp/asteroid/build/tmp-glibc/work/armv7ve-neon-oe-linux-gnueabi/qtwebkit/5.6.2+gitAUTOINC+bb43bca166-r0/temp/log.do_compile.17240

Log data follows:

| offlineasm: offset extractor LLIntDesiredOffsets.h successfully generated.
| compiling /tmp/asteroid/build/tmp-glibc/work/armv7ve-neon-oe-linux-gnueabi/qtwebkit/5.6.2+gitAUTOINC+bb43bca166-r0/git/Source/JavaScriptCore/llint/LLIntOffsetsExtractor.cpp
| In file included from /tmp/asteroid/build/tmp-glibc/sysroots/bass/usr/include/c++/6.2.0/bits/stl_algo.h:59:0,
|                  from /tmp/asteroid/build/tmp-glibc/sysroots/bass/usr/include/c++/6.2.0/algorithm:62,
|                  from /tmp/asteroid/build/tmp-glibc/sysroots/bass/usr/include/QtCore/qglobal.h:88,
|                  from /tmp/asteroid/build/tmp-glibc/work/armv7ve-neon-oe-linux-gnueabi/qtwebkit/5.6.2+gitAUTOINC+bb43bca166-r0/git/Source/WTF/wtf/Platform.h:1087,
|                  from /tmp/asteroid/build/tmp-glibc/work/armv7ve-neon-oe-linux-gnueabi/qtwebkit/5.6.2+gitAUTOINC+bb43bca166-r0/git/Source/JavaScriptCore/config.h:30,
|                  from /tmp/asteroid/build/tmp-glibc/work/armv7ve-neon-oe-linux-gnueabi/qtwebkit/5.6.2+gitAUTOINC+bb43bca166-r0/git/Source/JavaScriptCore/llint/LLIntOffsetsExtractor.cpp:26:
| /tmp/asteroid/build/tmp-glibc/sysroots/bass/usr/include/c++/6.2.0/cstdlib:75:25: fatal error: stdlib.h: No such file or directory
|  #include_next <stdlib.h>
|                          ^
| ERROR: oe_runmake failed
| compilation terminated.
| Makefile.LLIntOffsetsExtractor:408: recipe for target '.obj/llint/LLIntOffsetsExtractor.o' failed
| make[2]: *** [.obj/llint/LLIntOffsetsExtractor.o] Error 1
| make[2]: Leaving directory '/tmp/asteroid/build/tmp-glibc/work/armv7ve-neon-oe-linux-gnueabi/qtwebkit/5.6.2+gitAUTOINC+bb43bca166-r0/build/Source/JavaScriptCore'
| Makefile.JavaScriptCore:42: recipe for target 'sub-LLIntOffsetsExtractor-pro-make_first-ordered' failed
| make[1]: *** [sub-LLIntOffsetsExtractor-pro-make_first-ordered] Error 2
| make[1]: Leaving directory '/tmp/asteroid/build/tmp-glibc/work/armv7ve-neon-oe-linux-gnueabi/qtwebkit/5.6.2+gitAUTOINC+bb43bca166-r0/build/Source/JavaScriptCore'
| Makefile:92: recipe for target 'sub-Source-JavaScriptCore-JavaScriptCore-pro-make_first-ordered' failed
| make: *** [sub-Source-JavaScriptCore-JavaScriptCore-pro-make_first-ordered] Error 2
| WARNING: exit code 1 from a shell command.
| ERROR: Function failed: do_compile (log file is located at /tmp/asteroid/build/tmp-glibc/work/armv7ve-neon-oe-linux-gnueabi/qtwebkit/5.6.2+gitAUTOINC+bb43bca166-r0/temp/log.do_compile.17240)
ERROR: Task (/tmp/asteroid/src/meta-qt5/recipes-qt/qt5/qtwebkit_git.bb:do_compile) failed with exit code '1'
NOTE: Tasks Summary: Attempted 2360 tasks of which 1689 didn't need to be rerun and 1 failed.

Summary: 1 task failed:
  /tmp/asteroid/src/meta-qt5/recipes-qt/qt5/qtwebkit_git.bb:do_compile
Summary: There were 2 ERROR messages shown, returning a non-zero exit code.
```

parkerlreed, 2017-01-10:
If I may ask, how did you get bitbake working? I see it gets cloned as part of the initial pull but I never could get it setup correctly in my PATH.

Lrs121, 2017-01-10:
It gets temporarily added to the path when the prepare-build.sh script is run. If you run it with `. ./prepare-build.sh` or with `source ./prepare-build.sh` should be put into a directory called build. You call bitbake inside that folder and it should run.

FlorentRevest, 2017-01-10:
Hi aebruno,
The SDK build is indeed currently broken. You can find a workaround here: http://ronan.lepage1.free.fr/gumstix-wiki/doku.php?id=asteroidos:clean_start

aebruno, 2017-01-11:
Thanks @FlorentRevest for the pointer. I did a little more digging and was able to get it compiled with only 2 small patches to the recipes in meta-qt5. Here's a quick summary of what I did in case this may be of help to others.

The first issue seems to be related to how qmake passes some include paths with "-isystem" prefix [1,2]. This flag effects the internal header search path order and seems to cause issues when compiling the sdk. By default `QMAKE_CFLAGS_ISYSTEM = -isystem` see [3]. This patch resets `QMAKE_CFLAGS_ISYSTEM` to use `-I` instead. This cleared up all the `stdlib.h` not found errors and qtwebkit compiled cleanly. Here's the patch I applied to `src/meta-qt5/recipes-qt/qt5/qtwebkit_git.bb`:

```diff
--- a/recipes-qt/qt5/qtwebkit_git.bb
+++ b/recipes-qt/qt5/qtwebkit_git.bb
@@ -53,7 +53,7 @@ do_configure_prepend() {
 }
 
 # qtwebkit gets terribly big when linking with all debug info, disable by default
-QTWEBKIT_DEBUG = "QMAKE_CFLAGS+=-g0 QMAKE_CXXFLAGS+=-g0"
+QTWEBKIT_DEBUG = "QMAKE_CFLAGS+=-g0 QMAKE_CXXFLAGS+=-g0 QMAKE_CFLAGS_ISYSTEM="
 EXTRA_QMAKEVARS_PRE += "${QTWEBKIT_DEBUG}"
```
Next, added/removed the packages described in [4] from
`src/meta-qt5/recipes-qt/packagegroups/packagegroup-qt5-toolchain-target.bb`

```diff
--- a/recipes-qt/packagegroups/packagegroup-qt5-toolchain-target.bb
+++ b/recipes-qt/packagegroups/packagegroup-qt5-toolchain-target.bb
@@ -10,9 +10,6 @@ PACKAGEGROUP_DISABLE_COMPLEMENTARY = "1"
 # Requires Ruby to work
 USE_RUBY = " \
     qtquick1-dev \
-    qtquick1-mkspecs \
-    qtquick1-plugins \
-    ${@bb.utils.contains('DISTRO_FEATURES', 'opengl', 'qtquick1-qmlplugins', '', d)} \
     qttranslations-qtquick1 \
     qtwebkit-dev \
     qtwebkit-mkspecs \
@@ -36,9 +33,11 @@ USE_X11 = " \
 RDEPENDS_${PN} += " \
     packagegroup-core-standalone-sdk-target \
     libsqlite3-dev \
+    mapplauncherd-qt \
     ${@bb.utils.contains('DISTRO_FEATURES', 'opengl', 'qt3d-dev', '', d)} \
     ${@bb.utils.contains('DISTRO_FEATURES', 'opengl', 'qt3d-mkspecs', '', d)} \
     ${@bb.utils.contains('DISTRO_FEATURES', 'opengl', 'qt3d-qmlplugins', '', d)} \
+    qtquick1 \
     qtbase-dev \
     qtbase-fonts \
     qtbase-mkspecs \
@@ -72,7 +71,6 @@ RDEPENDS_${PN} += " \
     qttranslations-qtlocation \
     qtmultimedia-dev \
     qtmultimedia-mkspecs \
-    qtmultimedia-plugins \
     ${@bb.utils.contains('DISTRO_FEATURES', 'opengl', 'qtmultimedia-qmlplugins', '', d)} \
     qttranslations-qtmultimedia \
     qtscript-dev \
@@ -91,8 +89,6 @@ RDEPENDS_${PN} += " \
     qtsystems-mkspecs \
     ${@bb.utils.contains('DISTRO_FEATURES', 'opengl', 'qtsystems-qmlplugins', '', d)} \
     qttools-dev \
-    qttools-mkspecs \
-    qttools-staticdev \
     qttools-tools \
     ${@bb.utils.contains('DISTRO_FEATURES', 'wayland', '${USE_WAYLAND}', '', d)} \
     ${USE_RUBY} \
```

With the above 2 patches applied the sdk compiled successfully. Given that these issues are in meta-qt5 and live outside of AsteroidOS core, perhaps these patches could be incorporated into the `prepare-build.sh` script? If you think this would be a good place for it I'd be happy to provide a pull request.

Thanks for all the great work on AsteroidOS! 

[1] http://lists.openembedded.org/pipermail/openembedded-core/2016-August/125952.html
[2] http://stackoverflow.com/questions/37218953/isystem-on-a-system-include-directory-causes-errors
[3] http://code.qt.io/cgit/qt/qtbase.git/tree/mkspecs/common/gcc-base.conf?h=5.6#n47
[4] http://ronan.lepage1.free.fr/gumstix-wiki/doku.php?id=asteroidos:clean_start

FlorentRevest, 2017-01-11:
Good work! The only proper place to PR those modification is in a bbappend file in meta-asteroid.
Anyway, congratulations for your findings.

Florent

aebruno, 2017-01-11:
Great, I'll see if I can something added to meta-asteroid. I should have mentioned I did not test re-building the asteroid image and not sure if these changes would have a negative impact on that build.

Thanks!

closed 2017-01-11 by FlorentRevest

parkerlreed, 2017-01-13:
Quick question: 

````
source ./prepare-build.sh sparrow
````

Is not putting bitbake in my $PATH. Is it supposed to?

````
[parker@e55 build]$ git clone https://github.com/AsteroidOS/asteroid.git
Cloning into 'asteroid'...
remote: Counting objects: 108, done.
remote: Total 108 (delta 0), reused 0 (delta 0), pack-reused 108
Receiving objects: 100% (108/108), 24.18 KiB | 0 bytes/s, done.
Resolving deltas: 100% (62/62), done.
[parker@e55 build]$ cd asteroid/
[parker@e55 asteroid]$ source ./prepare-build.sh sparrow
Cloning into 'src/oe-core'...
remote: Counting objects: 272965, done.
remote: Compressing objects: 100% (71863/71863), done.
remote: Total 272965 (delta 196426), reused 272651 (delta 196138)
Receiving objects: 100% (272965/272965), 95.40 MiB | 2.77 MiB/s, done.
Resolving deltas: 100% (196426/196426), done.
Cloning into 'src/oe-core/bitbake'...
remote: Counting objects: 46977, done.
remote: Compressing objects: 100% (10421/10421), done.
remote: Total 46977 (delta 35205), reused 46855 (delta 35117)
Receiving objects: 100% (46977/46977), 10.03 MiB | 3.88 MiB/s, done.
Resolving deltas: 100% (35205/35205), done.
Cloning into 'src/meta-openembedded'...
remote: Counting objects: 69759, done.
remote: Compressing objects: 100% (120/120), done.
remote: Total 69759 (delta 95), reused 15 (delta 15), pack-reused 69624
Receiving objects: 100% (69759/69759), 28.12 MiB | 290.00 KiB/s, done.
Resolving deltas: 100% (39988/39988), done.
Cloning into 'src/meta-asteroid'...
remote: Counting objects: 2425, done.
remote: Compressing objects: 100% (7/7), done.
remote: Total 2425 (delta 0), reused 0 (delta 0), pack-reused 2417
Receiving objects: 100% (2425/2425), 353.24 KiB | 0 bytes/s, done.
Resolving deltas: 100% (1251/1251), done.
Cloning into 'src/meta-smartphone'...
remote: Counting objects: 22026, done.
remote: Compressing objects: 100% (150/150), done.
remote: Total 22026 (delta 103), reused 1 (delta 1), pack-reused 21873
Receiving objects: 100% (22026/22026), 8.28 MiB | 603.00 KiB/s, done.
Resolving deltas: 100% (9727/9727), done.
Cloning into 'src/meta-qt5'...
remote: Counting objects: 7094, done.
remote: Total 7094 (delta 0), reused 0 (delta 0), pack-reused 7094
Receiving objects: 100% (7094/7094), 2.04 MiB | 0 bytes/s, done.
Resolving deltas: 100% (4366/4366), done.
Cloning into 'src/meta-sparrow-hybris'...
remote: Counting objects: 385, done.
remote: Total 385 (delta 0), reused 0 (delta 0), pack-reused 385
Receiving objects: 100% (385/385), 221.91 KiB | 0 bytes/s, done.
Resolving deltas: 100% (128/128), done.
Cloning into 'src/meta-tetra-hybris'...
remote: Counting objects: 162, done.
remote: Total 162 (delta 0), reused 0 (delta 0), pack-reused 162
Receiving objects: 100% (162/162), 64.77 KiB | 0 bytes/s, done.
Resolving deltas: 100% (40/40), done.
Cloning into 'src/meta-bass-hybris'...
remote: Counting objects: 419, done.
remote: Total 419 (delta 0), reused 0 (delta 0), pack-reused 419
Receiving objects: 100% (419/419), 420.70 KiB | 0 bytes/s, done.
Resolving deltas: 100% (137/137), done.
Cloning into 'src/meta-dory-hybris'...
remote: Counting objects: 409, done.
remote: Compressing objects: 100% (3/3), done.
remote: Total 409 (delta 0), reused 0 (delta 0), pack-reused 406
Receiving objects: 100% (409/409), 447.92 KiB | 0 bytes/s, done.
Resolving deltas: 100% (134/134), done.
OpenEmbedded requires 'python' to be python v2 (>= 2.7.3), not python v3.
Please set up python v2 as your default 'python' interpreter.
Welcome to the Asteroid compilation script.

If you meet any issue you can report it to the project's github page:
    https://github.com/AsteroidOS

You can now run the following command to get started with the compilation:
    bitbake asteroid-image

Have fun!
[parker@e55 oe-core]$ bitbake
bash: bitbake: command not found
````

I haven't changed the python link yet but I don't think that would affect the script at least being in my $PATH.