AsteroidOS/asteroid #11: How to replace Android Wear?

mdibaiee, 2017-01-13:
Hey,
First, thanks for bringing this to us, great work indeed!

I'd like to know how can I replace Android Wear with Asteroid? I want to use Asteroid as my primary OS on my watch and I don't want to have to `fastboot boot` every time, I have the instructions to revert back to Android Wear just in case, though I don't think I ever will.

Thanks

parkerlreed, 2017-01-13:
Which watch? For my Zenwatch 2 (sparrow) I just took a backup of the stock boot partition and then flashed the Asteroid zImage with 

````
fastboot flash boot zImage-blah
````

I then made a backup of the Asteroid boot in TWRP so I can technically dual boot both. https://www.youtube.com/watch?v=6d0w_Q1IqGc

mdibaiee, 2017-01-13:
I have a LG G Watch (dory), I tried that, I would only get the "LG G Watch" screen and it would go blank afterwards. 🤔

FlorentRevest, 2017-01-13:
Really surprising. This has already been reported to work several times.
Maybe you'll have more chance flashing the recovery partition with the asteroid's boot.img? Actually I don't have many ideas on how to fix this behavior. The only explanation I can imagine would be an incorrect boot.img having been flashed or a missing rootfs.ext2 on the sdcard partition (you haven't wiped the android's data partition, have you?)

mdibaiee, 2017-01-14:
@FlorentRevest Oops, you were right, it seems I had wiped before doing this, sorry about that and thank you both for your help! Now it works like a charm, I'm excited to develop for Asteroid!

As a side note, I would recommend you mention this somewhere in case someone wants to do that.

FlorentRevest, 2017-03-29:
Some instructions about flashing asteroid on the userdata have been published on the Building AsteroidOS documentation. I close the issue

closed 2017-03-29 by FlorentRevest