AsteroidOS/asteroid #15: Back button on G Watch Urbane?

ericfont, 2017-03-25:
Sorry this might be an idiot question, but I just installed AsteroidOS Alpha 1 on G Watch Urbane (bass), but I can't figure out how back out of a menu.  For instance if I got to Settings->About, then I can't figure out how to get back to settings menu, no matter how I swipe or tap the screen.  There is only one phyiscal button on the watch, but all it does is turn the display off, and then when I press again, I'm back at home screen (clock).  But then going back to settings, it goes directly to About screen.  So I'm now unable to view anything else in the settings menu.

ericfont, 2017-03-25:
I just realized this repo is for the build system.  Sorry, but I don't know where to ask this question.

TheLastProject, 2017-03-25:
You should be able to swipe from the left edge to the right to go back a screen and from the top edge to the bottom to close an app

ericfont, 2017-03-25:
aha, it took me several attemps at swiping from left to right to get it to close.  I think part of the problem is that the G Watch Urbane has a circular display, so there is no straight left edge.  So maybe the circular edge is making it hard to detect a swipe from the very left bit of the edge.

ericfont, 2017-03-25:
just to note, I'm easily able to navigate left to right from the home screen using swipe (I don't have to make a full swipe from edge-to-edge, and I can make the swipe starting from any y position).  But when swiping to get out of the Settings About page, I have to carefully swipe from the left-most edge of the circle, and only at the middle y position).

ericfont, 2017-03-25:
I changed the title of this issue according to what I perceive the problem to be.

FlorentRevest, 2017-03-29:
Thanks for reporting, the experience should now be significantly better on every watches.

closed 2017-03-29 by FlorentRevest