AsteroidOS/asteroid #17: ERROR: Nothing PROVIDES dory

TheLastProject, 2017-03-30:
```
$ bitbake asteroid-image dory
Parsing recipes: 100% |################################################################################################################################################| Time: 0:03:50
Parsing of 2036 .bb files complete (0 cached, 2036 parsed). 2744 targets, 460 skipped, 0 masked, 0 errors.
ERROR: Nothing PROVIDES 'dory'

Summary: There was 1 ERROR message shown, returning a non-zero exit code.
```

Happens on both my Debian and Fedora after a successful ``. ./prepare-build.sh``, so seems to be something with the build scripts

FlorentRevest, 2017-03-30:
Why do you append "dory" to the bitbake command? The commands are

. ./prepare-build.sh dory
bitbake asteroid-image

https://asteroidos.org/wiki/building-asteroidos/

closed 2017-03-30 by FlorentRevest

DivyanshuSaxena, 2018-08-15:
The above two commands are not working for me. I built the dory image, using    
```. ./prepare-build.sh dory```. 
Here is my log.
```
debu@divy:~/asteroid$ . ./prepare-build.sh dory
Welcome to the Asteroid compilation script.

If you meet any issue you can report it to the project's github page:
    https://github.com/AsteroidOS

You can now run the following command to get started with the compilation:
    bitbake asteroid-image

Have fun!
debu@divy:~/asteroid/build$ bitbake asteroid-image
Loading cache: 100% |############################################| Time: 0:00:02
Loaded 3057 entries from dependency cache.
Parsing recipes: 100% |##########################################| Time: 0:00:36
Parsing of 2248 .bb files complete (2233 cached, 15 parsed). 3072 targets, 308 skipped, 0 masked, 0 errors.
NOTE: Resolving any missing task queue dependencies
ERROR: Nothing PROVIDES 'libhybris' (but /home/debu/asteroid/src/meta-asteroid/recipes-nemomobile/sensorfw/sensorfw_git.bb DEPENDS on or otherwise requires it)
libhybris was skipped: PREFERRED_PROVIDER_virtual/mesa set to mesa, not libhybris
NOTE: Runtime target 'sensorfw' is unbuildable, removing...
Missing or unbuildable dependency chain was: ['sensorfw', 'libhybris']
ERROR: Required build target 'asteroid-image' has no buildable providers.
Missing or unbuildable dependency chain was: ['asteroid-image', 'sensorfw', 'libhybris']

Summary: There were 2 ERROR messages shown, returning a non-zero exit code.
```

FlorentRevest, 2018-08-18:
@DivyanshuSaxena: Please open a new issue when you have a new problem.
I would advise to wipe your build/ directory and try again, there seems to be something wrong with your MACHINE configuration that makes bitbake think it is building for a qemu target instead of a watch target.