AsteroidOS/asteroid #18: Dory fails to build from source

cavokz, 2017-04-05:
Am I the only one unable to build Dory from a fresh checkout of today?

````
Summary: 5 tasks failed:
  /home/cavok/devel/asteroid.git/src/meta-asteroid/recipes-nemomobile/mlite/mlite_git.bb:do_compile
  /home/cavok/devel/asteroid.git/src/meta-asteroid/recipes-nemomobile/nemo-keepalive/nemo-keepalive_git.bb:do_compile
  /home/cavok/devel/asteroid.git/src/meta-asteroid/recipes-nemomobile/libsignon-qt5/libsignon-qt5_git.bb:do_compile
  /home/cavok/devel/asteroid.git/src/meta-asteroid/recipes-nemomobile/statefs/statefs-providers_git.bb:do_compile
  /home/cavok/devel/asteroid.git/src/meta-asteroid/recipes-asteroid/asteroid-btsyncd/asteroid-btsyncd_git.bb:do_compile
````

FlorentRevest, 2017-04-06:
Could you please try again and send complete paste of all the build errors? those few lines aren't enough for me to debug what the problem is. Note that asteroid-btsyncd should now be fixed.

cavokz, 2017-04-10:
Here is the log of this morning (it's the same build tree of the previous attempts, at the beginning I show it's updated): [asteroid.log.gz](https://github.com/AsteroidOS/asteroid/files/909378/asteroid.log.gz)

FlorentRevest, 2017-04-11:
The problem appears to come from meta-qt5. For some reason, their qtbase-native recipe compiles a segfaulting qdbusxml2cpp tool. The problem should be reported to their issues tracker.

If you want a quick and dirty fix you can replace your asteroid/build/tmp-glibc/sysroots/x86_64-linux/usr/bin/qdbusxml2cpp with http://placeholder.fr/~kido/qdbusxml2cpp then run bitbake -c cleanall mlite nemo-keepalive libsignon-qt5 statefs-providers

cavokz, 2017-04-11:
I'll check it out. I should receive my watch soon. Thanks.

cavokz, 2017-04-19:
I attached new logs: [dory.log.gz](https://github.com/AsteroidOS/asteroid/files/933045/dory.log.gz)
This is what I get after the clean you said but without replacing qdbusxml2cpp.
These are the errors I've got the very first time and I get them only if I do the clean. Maybe they are related?

cavokz, 2017-04-21:
Sorry, didn't realize it was the same problem. Anyway, your workaround worked, I've successful build. Thanks.

closed 2017-04-21 by cavokz