AsteroidOS/asteroid #19: help wanted

Caballlero, 2017-05-01:
HI 
i have installed asteroid os lg g watch yesterday first time, i tell you I REALLY LIKE  THIS OS.
- i have many questions i hope so you help respond.
- why dev build have ext.4 ? the official release is ext2 ?
- i tried install asteroid os with dev build ext4 example: adb push -p /home/carlos/Downloads/asteroid-image-dory.ext4 /sdcard/linux/rootfs.ext4 ,after next step fastboot boot /home/carlos/Downloads/zImage-dtb-dory.fastboot i do not get boot ?  so i changed the ext4 to ext2 using dev build after that BOOT AS WELL WITH ASTEROID . then i need help how to see or delete the rootfs.ext* created and start again with ext4 WITHOUT android wear ONLY ASTEROID OS !

FlorentRevest, 2017-05-01:
Hi,

I recently switched a couple of devices' preferred rootfs file formats to ext4 but the previous "/sdcard/linux/rootfs.ext2" path is still in use by the initramdisk. I admit there is an inconsistency here, I'll shortly move every device to ext4 and every rootfs to something like /sdcard/asteroidos.ext4 in the dev builds. I'll keep this issue opened while this hasn't been done.

If you want to delete your /sdcard/linux/rootfs.ext4 you should use "adb shell rm /sdcard/linux/rootfs.ext4" while running Android Wear.

If you want to install Asteroid on a partition and replace Android Wear, follow the last paragraph of https://asteroidos.org/wiki/building-asteroidos/

P.S: Just a couple of tips for your future contributions to open-source projects:
1- Please, use a clearer Issue title. You opened an "Issue", I *know* you have a problem, but "help wanted" doesn't explain anything. "Confusion between ext4 and ext2 rootfs on dory" could, for example, have been clearer.
2- No need to scream. https://www.lifewire.com/why-not-to-write-in-all-caps-1173242

If you are also "Cambot" on IRC (and sorry if you're not) I have two other tips for you:
1- Rule n°1 of IRC is "Don't ask to ask, just ask". We are here to talk, there's no need for "anyone online?" etc... http://sol.gfxile.net/dontask.html
2- Wait longer. We have lives too and aren't always looking at our IRC channels.

Caballlero, 2017-05-01:
- i already changed the title of this issue , the reason why i put "help wanted" it is because i have i lot of question(issues) of diferent titles. but apologize- thanks a lot  for your tips.

FlorentRevest, 2017-05-11:
Fixed since https://github.com/AsteroidOS/meta-asteroid/commit/e92aabcee7472ac97a0dda97b7ecd08872783f2e

closed 2017-05-11 by FlorentRevest