AsteroidOS/asteroid #20: Flash userdate  failed- dory

Caballlero, 2017-05-03:
HI florent
i tried install asteroid-only on my watch,after all my device not boot
root@debian-sony:/home/carlos# adb reboot bootloader
root@debian-sony:/home/carlos# fastboot flash userdata asteroid-image-dory.ext2target reported max download size of 268435456 bytes
Invalid sparse file format at header magi
erasing 'userdata'...
OKAY [  3.383s]
sending sparse 'userdata' 1/1 (212029 KB)...
OKAY [  6.781s]
writing 'userdata' 1/1...
FAILED (remote: sparse image write failure)
finished. total time: 73.893s
root@debian-sony:/home/carlos# fastboot flash boot zImage-dtb-dory.fastboot
target reported max download size of 268435456 bytes
sending 'boot' (17236 KB)...
OKAY [  0.663s]
writing 'boot'...
OKAY [  1.268s]
finished. total time: 1.930s

FlorentRevest, 2017-05-03:
Flashing the image on a partition can only be done with a development build. Currently, the dory development build is broken https://release.asteroidos.org/nightlies/dory/ check it out when it will be ready.

Caballlero, 2017-05-03:
ok. but my watch not have android wear . how can i   get fastboot mode to try again with con dev build ?

FlorentRevest, 2017-05-03:
Dev builds have been updated, you should be able to flash it.

You can enter bootloader mode by swiping your screen from top left to bottom right when the LG logo appears.

Caballlero, 2017-05-03:
- Thanks for the tips to get bootlader mode. 
-  I downloaded last dev build 03 May.
- i Flashed again > SUCCESS  :  )
#  FIXED

closed 2017-05-03 by FlorentRevest