AsteroidOS/asteroid #22: //  I do not find the string of word to translate

Caballlero, 2017-05-11:
-  I want to transate a spanish when plug or unplug the watch when is charging. only show in english by default. But I can not find these strings of words to translate. 

- thanks

FlorentRevest, 2017-05-11:
I don't understand the issue. Could you please clarify what you have in mind with examples of missing strings?

Caballlero, 2017-05-11:
- I call String Of Words like this,i looking this string on charging-usb section.
<?xml version="1.0" encoding="utf-8"?>
<!DOCTYPE TS>
<TS version="2.1" language="es">
<context>
    <name>AppSwitcher</name>
    <message>
        <location filename="../qml/AppSwitcher.qml" line="120"/>
        <source>Example</source>
        <translation>Example</translation>
    </message>
</context>
<context>
    <name>FeedsPage</name>
    <message>
        <location filename="../qml/FeedsPage.qml" line="119"/>
        <source>Example</source>
<translation>Example</translation>

Caballlero, 2017-05-12:
Did you understand me ?

FlorentRevest, 2017-05-13:
Thanks for your edits that made it much clearer. I have the issue in mind and am searching for a fix. I'll close the issue when solved.

FlorentRevest, 2017-05-14:
This is now fixed by https://github.com/AsteroidOS/lipstick/commit/83ffc2e9dbea566d007ba064099788c29bdb1610 and https://github.com/AsteroidOS/meta-asteroid/commit/82c60c399434309e889fcac8e50368bb083fbf9d

If you have changes to make to those strings, please contribute upstream at https://translate.sailfishos.org/es/lipstick-qt5/ I will not accept differences between these strings in AsteroidOS and in SailfishOS.

closed 2017-05-14 by FlorentRevest

Caballlero, 2017-05-14:
- thank you very much.
- I did changes in 83ffc2e  Before reading this sorry Florent. 
- I do  translate only sailfish apps, not sailfish upstream. I do not know what to do with sailfishOS, I do not know who is the SFOS Spanish translator and I do not want to modify that upstream because it is different from a cell phone screen to watch. What do you think?