AsteroidOS/asteroid #23: // asteroid not boot very well last dev-build

Caballlero, 2017-05-18:
After install dev-build 18-May-2017 asteroid like freeze in home. just accept unlock to home only. but slide to go apps settings not work, slide no work after flash.

Caballlero, 2017-05-18:
- This issue can not be solved  rebooting the watch : |

FlorentRevest, 2017-05-18:
Ok, this issue confirms what I suspected at 13:36:27 in https://log.asteroidos.org/?date=2017-05-16&level=16
Can you try the fix described in the message and tell me if 1- touchscreen is fixed after the scp and a reboot 2- vibrations work in alarms and timer.

Caballlero, 2017-05-18:
What is scp?
I can not slide to go a vibration or bluetooth options too.

FlorentRevest, 2017-05-18:
http://www.hypexr.org/linux_scp_help.php

Something like the following lines should do the trick.
wget https://raw.githubusercontent.com/AsteroidOS/meta-bass-hybris/474b28a601db6148f059cba144e8c9ee120aea80/recipes-asteroid/asteroid-launcher/asteroid-launcher/default.conf
scp default.conf root@192.168.2.15:/var/lib/environment/compositor/default.conf

Caballlero, 2017-05-19:
ok. i asked for a the last dev-build was 315MB now is over 400MB it is true the weight ?

FlorentRevest, 2017-05-19:
I don't understand what you mean with "true weight". Is this that unbelievable?
Replacing dropbear with openssh may have made the images larger and could explain your observation.

Caballlero, 2017-05-19:
If it is like saying it is normal to have over 400MB i the image? > OK i understand replace dropbear by openssh.
i do not know how to go  root@192.168.2.15 to patch ,because asteroidOS not boot anymore only show lg logo. Just report.

FlorentRevest, 2017-05-20:
Fixed by https://github.com/AsteroidOS/meta-dory-hybris/commit/b7a14dd0e498b63370bb960e2d4e9f351eb29973
Confirmed on IRC by Huntereb

closed 2017-05-20 by FlorentRevest