AsteroidOS/asteroid #24: Can't compile for tetra

RAZERZ, 2017-05-19:
Trying to compile for tetra on ubuntu 16.04 LTS, able to successfully compile for my phone (lineage and resurrection remix) but when I follow the steps on the website (https://asteroidos.org/wiki/building-asteroidos/) it all goes well till I run `bitbakte asteroid-image`. Here is the full output of what happens when I run it;

rami@Rami:~/android/asteroid/build$ bitbake asteroid-image tetra
ERROR:  OE-core's config sanity checker detected a potential misconfiguration.
    Either fix the cause of this error or at your own risk disable the checker (see sanity.conf).
    Following is the list of potential problems / advisories:

    Failed to create a file with a long name in TMPDIR. Please use a filesystem that does not unreasonably limit filename length.
Failed to create a file with a long name in SSTATE_DIR. Please use a filesystem that does not unreasonably limit filename length.

Summary: There was 1 ERROR message shown, returning a non-zero exit code.
rami@Rami:~/android/asteroid/build$ 

Was going to try to compile for dory instead, but I don't own that device so that would be such a waste of my time and resources. I can you with as much information as possible to resolve the issue :)

RAZERZ, 2017-05-19:
Oh and btw, I tried looking for the sanity.conf by doing find ./ sanity.conf in my ~/android/asteroid directory but it couldn't find it.

FlorentRevest, 2017-05-19:
Well, the error seems clear enough. It looks like you aren't using a "reasonable" filesystem.
I'm also afraid the error you meet is very tighly related to the way OpenEmbedded itself works, not AsteroidOS. So you are not reporting this "problem" to the right project.
Just one more thing, you seem to compare Asteroid's build system to Lineage's build system but those are two completely different build systems. Compiling Asteroid has nothing to do with compiling Android.

closed 2017-05-19 by FlorentRevest

RAZERZ, 2017-05-19:
What I meant that my build environment is set up for building. I'll look
more into what the error spit out.

1336117 RAZERZ SK488

Closed #24 <https://github.com/AsteroidOS/asteroid/issues/24>.

<https://github.com/AsteroidOS/asteroid/issues/24#event-1090203597>, or mute
the thread