AsteroidOS/asteroid #25: Bitbake exits with errors, sanity.conf is non-existent.

RAZERZ, 2017-05-20:
I have to do "touch conf/sanity.conf" to generate that file then when I run it it outputs a couple of errors then exits. Here is the pastebin for the full output (https://pastebin.com/RmR32b38) of what happens when I run bitbake. Running ubuntu 16.04 LTS, any help is appreciated ;)

TheLastProject, 2017-05-20:
You already got your answer in https://github.com/AsteroidOS/asteroid/issues/24, yet you refused to listen both to Bitbake and to @FlorentRevest.

If you look at the log, you clearly see it went wrong for the exact reason both Bitbake and @FlorentRevest told you it went wrong:
```
OSError: [Errno 36] File name too long: '/home/rami/android/asteroid/build/sstate-cache/22/sigtask.i3h7cnrs' -> '/home/rami/android/asteroid/build/sstate-cache/22/sstate:qtxmlpatterns:armv7ve-neon-oe-linux-gnueabi:5.8.0+gitAUTOINC+9f7e01b582:r0:armv7ve-neon:3:22479fa76a325d6ba0c0842691858910_generate_qt_config_file.tgz.siginfo'
```

You can say what you want, but your system is NOT set up correctly to compile AsteroidOS. Creating a sanity.conf to tell Bitbake "no, I know what I'm doing" doesn't magically make your system okay for building AsteroidOS on.

Use a system with "a file system that does not unreasonable limit filename length" like Bitbake told you or just give up on compiling and download a precompiled nightly from https://release.asteroidos.org/.

RAZERZ, 2017-05-20:
Yes but when I installed the packages that were in the guide in the website, about how to set up the environment, shouldn't that have done that? I did listen, to what @FlorentRevest said (or wrote) and I did some research on how long a filename could be on my os and it should be supported. But as you probably (I hope, since you starting bashing me on the one last thing at least) saw, that wasn't the only error I got, right? Good (or bad) so far but why do they show up?

RAZERZ, 2017-05-20:
Also, is it normal for sanity.conf not to be included or something?

RAZERZ, 2017-05-20:
Some more deep research shows that my home partition is encrypted using ecryptfs, which allows for a smaller name size. I should have paid a bit more attention to the video that was posted on your site because there it indicates that the user has root privilege and when I was root I could create this directory without any issue. Going to try to compile later on.

TheLastProject, 2017-05-20:
If your home partition is encrypted using ecryptfs, the easiest workaround would be to simply create a new user that doesn't use ecryptfs and compile it on that account. Doing things as root unnecessarily is generally a really bad idea. I wasn't aware of ecryptfs doing this, that's good to know (I only have experience with LVM).

RAZERZ, 2017-05-20:
Yeah I was doubtful when I said to myself that I have to compile in /root because the homepartition is locked, or I could just create a new directory in / but that wouldn't be so ideal since I compile for phones and you have to have your .profile and .bashrc and bin/ in your home partition. Well, it is worth a try.

FlorentRevest, 2017-05-21:
As noticed by bitbake, your filesystem wasn't adapted. This is not a bug either, I close the issue too.

closed 2017-05-21 by FlorentRevest

RAZERZ, 2017-05-21:
But where's the sanity.conf? Do I have to run touch conf/sanity.conf
everytime?

1336117 RAZERZ SK488

FlorentRevest, 2017-05-21:
The very first result when googling "sanity.conf" is https://github.com/openembedded/openembedded-core/blob/master/meta/conf/sanity.conf#L5
This file basically tells bitbake "Stop complaining about my filesystem, I really know what I am doing." and it really looks like you don't.

RAZERZ, 2017-05-21:
I don't now that's the thing now listen, when I ran without sanity.conf, it
wouldn't work, it'd just output and error that it wouldn't find the file
and stop. The sanity.conf is empty btw, so I tried with the touch command
and it worked :/

1336117 RAZERZ SK488

openembedded/openembedded-core/blob/master/meta/conf/sanity.conf#L5

TheLastProject, 2017-05-21:
Bitbake told you this in the original error message and I told you this in your second issue when you decided to ignore Bitbake by creating sanity.conf and ran into the exact issue Bitbake warned you about. Now @FlorentRevest is the third person to tell you yet you ask the exact same question again. At this point I genuinely wonder why you insist on asking for help yet won't listen to any answers. It's a waste of everyone's time, yours included, to do this.

I'll repeat what was said 3 times one last time: If you create a sanity.conf, you tell Bitbake: "don't check if things are okay, I know what I'm doing". A sanity.conf should normally NEVER exist.

RAZERZ, 2017-05-21:
I know that, but I'm still curious to why it asked me even when I switched
to another directory that's isn't encrypted and could handle a longer name

1336117 RAZERZ SK488

FlorentRevest, 2017-05-21:
Your build system may still be misconfigured with some references left to your ecryptfs partition.
If it's not the case let me tell you [once again](https://github.com/AsteroidOS/asteroid/issues/24#issuecomment-302809136) that you are reporting those errors to the wrong project. If bitbake isn't able to correctly identify your partition, we are not the developers that will be able to help you.

RAZERZ, 2017-05-21:
Yeyeye the important thing is learning for me (I completely removed the
asteroid got and fetched it in a non encrypted directory) and I believe
I've done that now

1336117 RAZERZ SK488