AsteroidOS/asteroid #26: Finished build is tiny (less than 100 bytes)

RAZERZ, 2017-05-20:
So the build finally compiled (after a couple of hours) but the end result is tiny!
I built for the sony smartwatch 3 (tetra) and I got the files that where available for download (except the one I made was ext4 instead of ext2) but the files on the website are around 300+ megabytes, while the ones I got in the end where tiny! Are they supposed to be like that? Thanks.

RAZERZ, 2017-05-20:
Update: I think I was talking about the wrong files... The ones I got had the same names etc but I found other files that were larger, they're rootfs though.

FlorentRevest, 2017-05-21:
The tiny files with short names are symbolic links to the larger files with names including build date. This is not a bug

closed 2017-05-21 by FlorentRevest

RAZERZ, 2017-05-21:
Running "ls -al" confirmed this

1336117 RAZERZ SK488