AsteroidOS/asteroid #35: Asteroid OS not boot after flash then choosing language

Caballlero, 2017-08-02:
After flash then choosing language the watch need reboot to make effect the language then most time not boot as well.

Caballlero, 2017-08-07:
Hi 
Any info of this?

FlorentRevest, 2017-08-07:
No update on this. This is most likely a corrupted partition which seems to happen quite often now for no apparent reason. That would need further debugging from the ramdisk but I can not reproduce the issue myself and I don't think I can tele-guide you through the debugging of this...

Caballlero, 2017-08-07:
Ok. What would be a simple solution on this?

FlorentRevest, 2017-08-07:
Not all the problems have simple solutions!

Caballlero, 2017-08-07:
:  )  Is this caused by making  many flash on the watch?

FlorentRevest, 2017-08-07:
It can be one of the cause. Also, do you force your watch to shutdown by holding the power button for a while when you reboot after selecting a new language? Because this can lead to fs inconsistencies. Fsck is ran at each boot in order to fix this kind of problems but sometimes the fs is in such a bad state that it can't be fixed.

Caballlero, 2017-08-07:
I do some time, but apparently only happens when i select a language, because after flash the clock, the clock turns on well. The problem is choosing a language which the clock needs to restart to take effect, where it is locked with white screen . . .

Caballlero, 2017-08-07:
Also i am compiling the source(AndroidSinc) ,but i get this error . . .
FAILURE: Build failed with an exception.

* What went wrong:
Execution failed for task ':app:lint'.
> Lint found errors in the project; aborting build.
  
  Fix the issues identified by lint, or add the following to your build script to proceed with errors:
  ...
  android {
      lintOptions {
          abortOnError false
      }
  }
  ...

* Try:

Caballlero, 2017-08-07:
I fixed this 
Fix the issues identified by lint, or add the following to your build script to proceed with errors:
...
android {
lintOptions {
abortOnError false
}

Caballlero, 2017-08-09:
I flashed today my watch , it is look like fixed this issue.

closed 2017-08-09 by Caballlero