AsteroidOS/asteroid #37: bootloader for LG Urbane 2 "nemo"

dk14, 2017-09-06:
Would be nice to have boot-loader image for LG nemo watch.

theappleman, 2017-09-06:
Hi @dk14,

while work on the nemo image has started, the image didn't boot properly so it has not been exposed in the prepare-build.sh script. Some additional testing with the initramfs needs to be carried out.

For reference, I've put the work [here](https://github.com/theappleman/meta-bass-hybris/tree/nemo), [here](https://github.com/theappleman/asteroid) and [here](https://github.com/theappleman/meta-asteroid/tree/libhybris-nemo); although the branches are a bit outdated, nemo has a lot in common with the LG watch urbane (bass) and other LG watches and would just require some adjustments to the dory/bass patches to bring it up to date.

dk14, 2017-09-07:
Oh thanks! I'll try to play with it. 

A bit unrelated question: the absence of sensors (particularly gyro and heart-rate) support for "bass"/"nemo" is just in-development or is there any legal/proprietary restrictions/requirements that don't allow it to be implemented (in open source)? 

Also, (I'm a noob in this) what low-level libraries/drivers could be used in order to access let's say heart-rate sensor (maybe not hrt itself but raw light-intensity data)?

theappleman, 2017-09-07:
There is nothing that I'm aware of stopping us from getting sensors working except that it needs to be looked in to. Dory has working sensors, although that sensor package is different to the ones in bass/nemo.

Information about sensors can be found here https://asteroidos.org/wiki/porting-guide/#sensors

MagneFire, 2020-12-01:
Someone on IRC asked for a build for `nemo`. I went ahead and picked up your work.
I don't know the exact state of this port. All I did was:
- Use the kernel recipes from @theappleman located here https://github.com/theappleman/meta-bass-hybris/tree/nemo/recipes-kernel/linux
  - Added some more patches to backport the Bluetooth driver.
- Rebuild system tar from scratch, original link in repo is dead.
- Use different partition names as provided by the person from IRC.

A nightly is available here:
https://www.dropbox.com/sh/am8t1ckpg5uvjir/AACANmCnFdIXFhj0QZ4CdgdEa?dl=0

Not sure if it works at all, but if someone wants to help with a port for this, then you are welcome :smile:

beroset, 2023-02-09:
Yay!  We now have this!  https://github.com/AsteroidOS/meta-smartwatch/tree/master/meta-nemo

eLtMosen, 2023-02-09:
Closing this issue and invite everyone to open new issues regarding missing features in meta-smartwatch repo tagged with `nemo`

closed 2023-02-09 by eLtMosen