AsteroidOS/asteroid #38: Cannot asteroid on sony smartwatch 3

themester, 2017-09-12:
mester > sudo fastboot flash userdata asteroid-image-tetra.ext4 
target reported max download size of 425721856 bytes
Invalid sparse file format at header magi
erasing 'userdata'...
OKAY [  9.256s]
sending sparse 'userdata' 1/2 (410009 KB)...
OKAY [ 18.367s]
writing 'userdata' 1/2...
FAILED (remote: unknown chunk type in sparse image)
finished. total time: 27.655s

comradekingu, 2017-09-12:
Did you checksum the downloaded image?

FlorentRevest, 2017-09-12:
This is a duplicate of https://github.com/AsteroidOS/asteroid/issues/31 and a known bug. I close this issue, please follow the other thread.

closed 2017-09-12 by FlorentRevest

themester, 2017-09-13:
@FlorentRevest I readed the other issue, and don't explain the solution