AsteroidOS/asteroid #39: [Audio] Notification Sound

FlorentRevest, 2017-10-04:
Some watches include a speaker (such as the prototype of Connect Watch), and can trigger sounds on notification events. A short alarm sound like this one: https://www.zapsplat.com/music/app-game-or-system-interface-alert-tone-38/ would be very helpful to make watch notifications more noticeable.

A short sound (in wav format) should be designed for AsteroidOS and made available for all under a permissive license. The best would be to actually make some kind of sheet music or midi file available for anyone to edit the file.

eLtMosen, 2022-08-06:
As recently discussed on matrix, we look for sounds for the following events:

- Timer finished
- Alarm Clock event
- Calendar event
- Startup, very subtile, accompanying the visual swooch when trasitioning from bootsplash to homescreen.
- System shut down ✅
- New Notification ✅
- Incoming call

After playing with LMMS 1.2.2, following sound design found some consent:
Notification sound:
[aos-notif-4-1-500ms.zip](https://github.com/AsteroidOS/asteroid/files/9275331/aos-notif-4-1-500ms.zip)

System shut down:
[aos-shutdown-4.zip](https://github.com/AsteroidOS/asteroid/files/9275332/aos-shutdown-4.zip)

LMMS project files:
[aos-sounds-lmms1-2-2.zip](https://github.com/AsteroidOS/asteroid/files/9275333/aos-sounds-lmms1-2-2.zip)

eLtMosen, 2023-03-24:
The audio files have been added to the new asteroid-media repo
https://github.com/AsteroidOS/asteroid-media/pull/9

eLtMosen, 2023-04-25:
We decided to add a mute button before enabling the sound deployment
https://github.com/AsteroidOS/asteroid-settings/issues/76