AsteroidOS/asteroid #40: Unable to build AsteroidOS

craftyguy, 2017-10-05:
Full output after following [instructions here](https://asteroidos.org/wiki/building-asteroidos/):

```
(venv-2) [clayton@computer build]$ bitbake asteroid-image                                 
/media/data/clayton/src/asteroid/build/tmp-glibc/hosttools/gcc: line 3: /media/data/clayton/src/asteroid/build/tmp-glibc/hosttools/../common.sh: No such file or directory:  0:00:47
/media/data/clayton/src/asteroid/build/tmp-glibc/hosttools/gcc: line 79: run_wrapped_binary: command not found                                                                      
/media/data/clayton/src/asteroid/build/tmp-glibc/hosttools/gcc: line 3: /media/data/clayton/src/asteroid/build/tmp-glibc/hosttools/../common.sh: No such file or directory          
ERROR: /media/data/clayton/src/asteroid/src/meta-smartphone/meta-android/recipes-core/images/chroot-image.bb: Error running gcc  --version:                                         
WARNING: /media/data/clayton/src/asteroid/src/meta-smartphone/meta-android/recipes-core/images/chroot-image.bb: Exception during build_dependencies for create_shar                 
WARNING: /media/data/clayton/src/asteroid/src/meta-smartphone/meta-android/recipes-core/images/chroot-image.bb: Error during finalise of /media/data/clayton/src/asteroid/src/meta-s
martphone/meta-android/recipes-core/images/chroot-image.bb                                
/media/data/clayton/src/asteroid/build/tmp-glibc/hosttools/gcc: line 79: run_wrapped_binary: command not found                                                                      
ERROR: ExpansionError during parsing /media/data/clayton/src/asteroid/src/meta-smartphone/meta-android/recipes-core/images/chroot-image.bb                                          
Traceback (most recent call last):           
bb.data_smart.ExpansionError: Failure expanding variable create_shar, expression was    # copy in the template shar extractor script                                                
        cp /media/data/clayton/src/asteroid/src/oe-core/meta/files/toolchain-shar-extract.sh /media/data/clayton/src/asteroid/build/tmp-glibc/work/dory-oe-linux-gnueabi/chroot-imag
e/1.0-r0/x86_64-deploy-chroot-image-populate-sdk/oecore-x86_64-armv7vehf-neon-toolchain-nodistro.0.sh                                                                               

        rm -f /media/data/clayton/src/asteroid/build/tmp-glibc/work/dory-oe-linux-gnueabi/chroot-image/1.0-r0/temp/pre_install_command /media/data/clayton/src/asteroid/build/tmp-gl
ibc/work/dory-oe-linux-gnueabi/chroot-image/1.0-r0/temp/post_install_command              

        if [ 1 -eq 1 ] ; then                
                cp /media/data/clayton/src/asteroid/src/oe-core/meta/files/toolchain-shar-relocate.sh /media/data/clayton/src/asteroid/build/tmp-glibc/work/dory-oe-linux-gnueabi/ch
root-image/1.0-r0/temp/post_install_command  
        fi                                   
        cat << "EOF" >> /media/data/clayton/src/asteroid/build/tmp-glibc/work/dory-oe-linux-gnueabi/chroot-image/1.0-r0/temp/pre_install_command                                    

EOF                                          

        cat << "EOF" >> /media/data/clayton/src/asteroid/build/tmp-glibc/work/dory-oe-linux-gnueabi/chroot-image/1.0-r0/temp/post_install_command                                   

EOF                                          
        sed -i -e '/@SDK_PRE_INSTALL_COMMAND@/r /media/data/clayton/src/asteroid/build/tmp-glibc/work/dory-oe-linux-gnueabi/chroot-image/1.0-r0/temp/pre_install_command' \         
                -e '/@SDK_POST_INSTALL_COMMAND@/r /media/data/clayton/src/asteroid/build/tmp-glibc/work/dory-oe-linux-gnueabi/chroot-image/1.0-r0/temp/post_install_command' \      
                /media/data/clayton/src/asteroid/build/tmp-glibc/work/dory-oe-linux-gnueabi/chroot-image/1.0-r0/x86_64-deploy-chroot-image-populate-sdk/oecore-x86_64-armv7vehf-neon
-toolchain-nodistro.0.sh                     

        # substitute variables               
        sed -i -e 's#@SDK_ARCH@#x86_64#g' \  
                -e 's#@SDKPATH@#/usr/local/oecore-x86_64#g' \                             
                -e 's#@SDKEXTPATH@#~/asteroid_sdk#g' \                                    
                -e 's#@OLDEST_KERNEL@#2.6.32#g' \                                         
                -e 's#@REAL_MULTIMACH_TARGET_SYS@#armv7vehf-neon-oe-linux-gnueabi#g' \    
                -e 's#@SDK_TITLE@#${@d.getVar("SDK_TITLE").replace('&', '\&')}#g' \       
                -e 's#@SDK_VERSION@#nodistro.0#g' \                                       
                -e '/@SDK_PRE_INSTALL_COMMAND@/d' \                                       
                -e '/@SDK_POST_INSTALL_COMMAND@/d' \                                      
                -e 's#@SDK_GCC_VER@#${@oe.utils.host_gcc_version(d)}#g' \                 
                /media/data/clayton/src/asteroid/build/tmp-glibc/work/dory-oe-linux-gnueabi/chroot-image/1.0-r0/x86_64-deploy-chroot-image-populate-sdk/oecore-x86_64-armv7vehf-neon
-toolchain-nodistro.0.sh                     

        # add execution permission           
        chmod +x /media/data/clayton/src/asteroid/build/tmp-glibc/work/dory-oe-linux-gnueabi/chroot-image/1.0-r0/x86_64-deploy-chroot-image-populate-sdk/oecore-x86_64-armv7vehf-neo
n-toolchain-nodistro.0.sh                    

        # append the SDK tarball             
        cat /media/data/clayton/src/asteroid/build/tmp-glibc/work/dory-oe-linux-gnueabi/chroot-image/1.0-r0/x86_64-deploy-chroot-image-populate-sdk/oecore-x86_64-armv7vehf-neon-too
lchain-nodistro.0.tar.xz >> /media/data/clayton/src/asteroid/build/tmp-glibc/work/dory-oe-linux-gnueabi/chroot-image/1.0-r0/x86_64-deploy-chroot-image-populate-sdk/oecore-x86_64-ar
mv7vehf-neon-toolchain-nodistro.0.sh         

        # delete the old tarball, we don't need it anymore                                
        rm /media/data/clayton/src/asteroid/build/tmp-glibc/work/dory-oe-linux-gnueabi/chroot-image/1.0-r0/x86_64-deploy-chroot-image-populate-sdk/oecore-x86_64-armv7vehf-neon-tool
chain-nodistro.0.tar.xz                      
 which triggered exception BBHandledException:                                            

ERROR: /media/data/clayton/src/asteroid/src/meta-smartphone/meta-android/recipes-core/images/initramfs-android-image.bb: Error running gcc  --version:                              
WARNING: /media/data/clayton/src/asteroid/src/meta-smartphone/meta-android/recipes-core/images/initramfs-android-image.bb: Exception during build_dependencies for create_shar      
WARNING: /media/data/clayton/src/asteroid/src/meta-smartphone/meta-android/recipes-core/images/initramfs-android-image.bb: Error during finalise of /media/data/clayton/src/asteroid
/src/meta-smartphone/meta-android/recipes-core/images/initramfs-android-image.bb          

Summary: There were 4 WARNING messages shown.                                             
Summary: There were 3 ERROR messages shown, returning a non-zero exit code.               
```

FlorentRevest, 2017-11-03:
Hi craftyguy,
Does this bug still happen? We have had several users successfully building AsteroidOS for the past month. I can't reproduce your problem either.
Since this is probably related to your host system, if the problem persists, could you please try to build Asteroid from an Ubuntu LTS LXC (or any distribution that is reasonably tested)

FlorentRevest, 2017-11-11:
Hi craftyguy,
Without answers from you and being able to reproduce the problem, I close the issue. Reopen with more details if it still happens

closed 2017-11-11 by FlorentRevest