AsteroidOS/asteroid #42: Add support for Huawei watches

grumpydev-net, 2017-10-24:
Self explanatory

FlorentRevest, 2017-11-03:
Afaik, no AsteroidOS developer owns a Huawei watch. Those watches are expensive and will probably take time before benefiting from an Asteroid port due to limited resources on our side.

However, on a purely technical point of view, the ports would be possible.

logic, 2017-11-04:
Are there any core devs in the SF bay area that'd be interested in borrowing my sturgeon, at least for the initial bootstrapping work (past what @bencord0 did initially)? It's currently just sitting in a drawer (I've since moved to a phone without google play services, so no android wear support), so it might as well get some use as a test mule. :wink:

(@FlorentRevest, I'd also be comfortable shipping it your way if it's something you'd be interested in working on, but I understand the hassle of shipping back and forth between the US and France might be more of a pain than you're interested in.)

FlorentRevest, 2017-11-04:
Hi @logic,

Thanks for the offer, this is very nice of you.
I already did a similar thing with another community member (for anthias) but the shipping was much easier (and cheaper) within our old continent. Moreover, I have lots of higher priority tasks I would want to work on first.

I'm not aware of any asteroid dev (or anyone with the time/willingness to learn) in the bay area but, if any, that would certainly be a simpler solution.

logic, 2017-11-04:
Perfectly fair. If I get a few cycles, I'll take another stab at getting a build up and running and see how far I can get with things, but I'm in the same boat (higher-priority stuff keeps pushing this down the list for me).

DarioXDF, 2019-03-15:
I would like to know if there is anything new for this topic since the last comment if from 2017 I have a Huawei GT smartwatch and I hate the Os of it

FlorentRevest, 2019-03-15:
Still no developers own a huawei watch so there hasn't been anything new on this topic since my last comment.

FlorentRevest, 2020-01-18:
Sturgeon support is now added to AsteroidOS.

closed 2020-01-18 by FlorentRevest