AsteroidOS/asteroid #43: Does not boot on Sony SmartWatch 3

Rudloff, 2017-11-01:
Hello,

I tried a temporary installation and after `fastboot boot zImage-dtb-tetra.fastboot` it just gets stuck on the "SmartWatch 3" boot screen for several minutes then reboots into Android Wear's charging mode.

FlorentRevest, 2017-11-03:
Hi Rudloff,

Thanks for reporting, I'll try to give this a look during the week-end.

gmelchett, 2017-11-03:
I have the same issue on my smartwatch 3 (build from last night), except that the watch doesn't reboot after a few minutes.
I can ssh to the watch, and looking at the output of: journalctl --no-pager,
this looks like the fault:
--
Nov 03 10:25:47 tetra systemd[500]: Failed to start The wayland compositors and desktop of AsteroidOS.
Nov 03 10:25:47 tetra systemd[500]: Dependency failed for Application launch booster for Qt Components on QtQuick2.
Nov 03 10:25:47 tetra systemd[500]: booster-qtcomponents-qt5.service: Job booster-qtcomponents-qt5.service/start failed with result 'dependency'.
Nov 03 10:25:47 tetra systemd[500]: asteroid-launcher.service: Unit entered failed state.
Nov 03 10:25:47 tetra systemd[500]: asteroid-launcher.service: Failed with result 'core-dump'.
Nov 03 10:25:47 tetra systemd[500]: Reached target Default.
--

There are lots of repeating pulseaudio issues before and after, but I guess those are irrelevant.

FlorentRevest, 2017-11-03:
Thank you gmelchett,
This sheds some light on the possible issue. I'll keep investigating.

FlorentRevest, 2017-11-04:
https://github.com/AsteroidOS/meta-tetra-hybris/commit/58589a87d4a35d70d31fba6b5e9985c0157ac31e fixed the problem reported by gmelchett.

The problem you reported, Rudloff, was probably temporary, I can't be reproduce it. Try next nightly build and reopen the issue if it still happens on your watch with more details on the commands you use.

closed 2017-11-04 by FlorentRevest

Rudloff, 2017-11-05:
I just tried the latest nightly and it booted correctly!