AsteroidOS/asteroid #45: Doubled notification

Etua, 2017-11-13:
_Tetra nightly image from 13.11.2017_
I got a message which resulted in showing some notification from Conversations, then I got an e-mail. After that I've noticed third notification on my SW3, it was a duplicate of the first one. Powering screen off and turning it on again resolved the problem.

FlorentRevest, 2018-03-12:
Could you please try again with AsteroidOSSync 0.9 ? It should be available on F-Droid in the upcoming days.

Etua, 2018-04-30:
@FlorentRevest I'll give it a try and close the issue if I won't encounter any problems.

Etua, 2020-01-30:
I don't remember whether the problem got solved but this issue is more than two years old now and tetra is not really supported anymore so in my opinion it does not make any sense to hold this thread open.

closed 2020-01-30 by Etua