AsteroidOS/asteroid #49: WebView?

Rudloff, 2017-12-07:
Hello,
Do you think it would be possible to have a webview in AsteroidOS?
This would allow us to port HTML/Cordova apps.

I tried using the WebView QML type but it does not appear to be supported:
```
qrc:/main.qml:22:1: module "QtWebView" is not installed
```

FlorentRevest, 2017-12-07:
Hi Rudloff.

If you need a web engine in your AsteroidOS build, you can easily add "qtwebengine" to your meta-asteroid/classes/asteroid-image.bbclass's IMAGE_INSTALL variable. If you have a recipe for an app that would require qtwebengine, you could also just add qtwebengine to its DEPENDS variable.

I decided it would be better to keep asteroid without a webengine because it takes a lot of space in the rootfs. However, if it ever gets well justified (for instance, if you have examples of Cordova apps that would be helpful to have in asteroid) that could be possible.

I close the issue for now.

closed 2017-12-07 by FlorentRevest

Rudloff, 2017-12-07:
I don't have a specific app in mind. It's just that there is a lot of web developers out there and I feel like this would provide a lower barrier to entry for porting or creating new apps.
But I understand your concerns.