AsteroidOS/asteroid #53: Cadran close Settings App

sheychen290, 2018-01-29:
In Settings App,
select "Cadran" menu close it after a frame on cadran selection page
and go back to app list.

swift 29/01/08 nightly build

FlorentRevest, 2018-01-29:
Can you please provide logs from journalctl -u user@1000 --no-pager when this bug happens?

sheychen290, 2018-01-29:
At 29 20:24:30

`Jan 29 20:23:57 swift asteroid-launcher[605]: Wayland display got fatal error 32: Broken pipe
Jan 29 20:23:57 swift asteroid-launcher[605]: Additionally, errno was set to 32: Broken pipe
Jan 29 20:23:57 swift asteroid-launcher[605]: The display is now unusable, aborting.
Jan 29 20:23:57 swift mapplauncherd[690]: Boosted process (pid=1896) was terminated due to signal 6
Jan 29 20:23:57 swift invoker[1949]: error: Can't send signal 6 to application [1896]: No such process
Jan 29 20:24:00 swift asteroid-launcher[605]: unsleepDisplay
Jan 29 20:24:06 swift asteroid-launcher[605]: sleepDisplay
Jan 29 20:24:24 swift asteroid-launcher[605]: unsleepDisplay
Jan 29 20:24:26 swift mapplauncherd[1950]: Couldn't open '/proc/self/oom_score_adj' for writing
Jan 29 20:24:27 swift asteroid-launcher[605]: Using the 'wl-shell' shell integration
Jan 29 20:24:27 swift booster-qtcomponents-qt5[690]: Using Wayland-EGL
Jan 29 20:24:30 swift asteroid-launcher[605]: file:///usr/share/asteroid-launcher/watchfaces/005-scifi.qml:73:5: QML Connections: Cannot assign to non-existent property "onChangesObserverChanged"
Jan 29 20:24:30 swift asteroid-launcher[605]: file:///usr/share/asteroid-launcher/watchfaces/004-alternative-digital-2.qml:169:5: QML Connections: Cannot assign to non-existent property "onChangesObserverChanged"
Jan 29 20:24:30 swift asteroid-launcher[605]: QWaylandGLContext::makeCurrent: eglError: 3009, this: 0x46bb18
Jan 29 20:24:30 swift asteroid-launcher[605]: Can't make current GL context
Jan 29 20:24:30 swift asteroid-launcher[605]: QOpenGLTexturePrivate::destroy() called without a current context.
Jan 29 20:24:30 swift asteroid-launcher[605]: Texture has not been destroyed
Jan 29 20:24:30 swift asteroid-launcher[605]: QOpenGLTexturePrivate::destroy() called without a current context.
Jan 29 20:24:30 swift asteroid-launcher[605]: Texture has not been destroyed
Jan 29 20:24:30 swift mapplauncherd[690]: Boosted process (pid=1950) was terminated due to signal 11
Jan 29 20:24:30 swift invoker[2073]: error: Can't send signal 11 to application [1950]: No such process`

Thanks

FlorentRevest, 2018-01-29:
Thanks for reporting.

FlorentRevest, 2018-02-19:
Closed by https://github.com/AsteroidOS/asteroid/commit/f20567e9568e1921f5f5ff904f80af705035f748

closed 2018-02-19 by FlorentRevest