AsteroidOS/asteroid #54: System requirements

Disctanger, 2018-02-09:
Good day AsteroidOS developers!
I am really amazed with your work and would love to contribute with porting. Could you please inform about system requirements for the OS so we can choose right device for porting. I don't have an experience in porting but want to learn. 
Currently I am interested in Diggro smartwaches (https://www.diggro.com/collections/all) and specifications of some of the products seems to be quite good.

FlorentRevest, 2018-02-09:
Hi dicstanger!

I'm glad you like Asteroid. :) Make sure you've given a look at https://asteroidos.org/wiki/porting-guide/ and https://asteroidos.org/wiki/porting-status/ before porting.

Diggro smartwatches are one of those cheap chinese Mediatek watches OEM. There are *tons* of them in Shenzhen doing pretty much the same watches, with a slightly different visual appearance. However, the specs are always ~the same derived from MTK reference devices. ( https://www.youtube.com/watch?v=vq7XjTYE5bc ) They usually have two lines of devices:
- Cheap (sometimes... *very cheap* https://www.youtube.com/watch?v=OGfOmbzinc0 ) watches running a real time operating system named "Nucleus" on a Cortex M microcontroller. Like the "Diggro GV68". Unfortunately, those watches can not run Asteroid because of hardware limitations. Their microcontrollers can not run a fully featured operating system with a kernel like Linux.
- "Expensive" Android watches like the "Diggro DI01". Those devices are much more capable and their hardware *could* run AsteroidOS.

While working on the "Connect Watch" ( https://www.ulule.com/connect-watch/ ), I was working on one of those typical Mediatek Android reference watch. It was running on a "MTK6580", the same SoC that you can find on the Diggro DI01. I got a viable proof of concept running so it would definitely be possible to make Asteroid run on those watches.
However, this required access to the watch's kernel's source code and to the android's source tree. And that's where it gets ugly. Because of various chinese legal bullshit, OEMs *sell* the access to their source code. (yes they should distribute the kernel of the GPL kernel for free, yes that's illegal, yes they don't give a darn)

Long story short, this means that if you are not working with Diggro themselves, and if you are not ready to put money on the table to buy something that should otherwise be freely available, you won't have access to the kernel source tree and you won't be able to port asteroid to those watches.

I would advise you to stick to the well paved way of Qualcomm watches. Qualcomm is not perfect either but at least they respect the GPL.

Disctanger, 2018-02-09:
@FlorentRevest 
Thanks for very informative response. 
I will try to contact Diggro guys and get more info about their watch. Moreover I will checkout Qualcomm watches

FlorentRevest, 2018-02-09:
You're very welcome. Good luck with contacting them !
Feel free to comment here, I just closed the issue so that it doesn't appear as "a bug".

closed 2018-02-09 by FlorentRevest