AsteroidOS/asteroid #55: qtbase-native build error

Rudloff, 2018-02-17:
Hello,

I'm trying to run `bitbake asteroid-image` in 2dddb62 and I get this error:
```
Loading cache: 100% |##########################################################################################################################################################| Time: 0:00:00
Loaded 3046 entries from dependency cache.
Parsing recipes: 100% |########################################################################################################################################################| Time: 0:00:06
Parsing of 2240 .bb files complete (2226 cached, 14 parsed). 3060 targets, 308 skipped, 0 masked, 0 errors.
NOTE: Resolving any missing task queue dependencies

Build Configuration:
BB_VERSION           = "1.36.0"
BUILD_SYS            = "x86_64-linux"
NATIVELSBSTRING      = "debian-9"
TARGET_SYS           = "arm-oe-linux-gnueabi"
MACHINE              = "tetra"
DISTRO               = "asteroid"
DISTRO_VERSION       = "alpha-1.0"
TUNE_FEATURES        = "arm armv7ve vfp thumb neon callconvention-hard"
TARGET_FPU           = "hard"
meta-qt5             = "5.10:0924e7a15d9ce36ef1869175cae3738071bd9edb"
meta                 = "rocko:abc7fbab2c125b247f3274ca4b29e4f858b5ae12"
meta-asteroid        = "master:3c49fd588bf8743cc93e5e88edba1ffb02602b1d"
meta-oe              
meta-multimedia      
meta-gnome           
meta-networking      = "rocko:a65c1acb1822966c3553de9fc98d8bb6be705c4e"
meta-android         = "rocko:4dc893c799951d8a9418860a6db9b3b7eed7bb49"
meta-python          
meta-filesystems     = "rocko:a65c1acb1822966c3553de9fc98d8bb6be705c4e"
meta-anthias-hybris  = "master:7e45263168efcb1696732b70204bd4fc6113c8d6"
meta-sparrow-hybris  = "master:55f5c9daf3e8e0797d9f49312e71e245f010fbe6"
meta-sprat-hybris    = "master:e64adbc6db63f7f1a90f0bdfaf8db123c3ac4a02"
meta-tetra-hybris    = "master:014b3436b930831dfaa5dd37510ebe7ac4944325"
meta-bass-hybris     = "master:248c7d5431ee2b540edbd5828ee125947f6b0de5"
meta-dory-hybris     = "master:cfa0f5c1c92dece99060c96da06f422c519a6d0b"
meta-lenok-hybris    = "master:462f80f0c45c8fdb4c37f943fbe8e3145324f8b8"
meta-swift-hybris    = "master:afd2e55dd62ef154e07afe66ee49418b51338314"
meta-wren-hybris     = "master:ef7d856fa95791bc03190ad2db704490c4ef96b1"

Initialising tasks: 100% |#####################################################################################################################################################| Time: 0:00:06
NOTE: Executing SetScene Tasks
NOTE: Executing RunQueue Tasks
ERROR: qtbase-native-5.10.1+gitAUTOINC+6c6ace9d23-r0 do_configure: Configuring qt failed. PACKAGECONFIG_CONFARGS was      -sysroot /home/pierre/www/asteroid/build/tmp-glibc/work/x86_64-linux/qtbase-native/5.10.1+gitAUTOINC+6c6ace9d23-r0/recipe-sysroot-native     -no-gcc-sysroot     -system-zlib     -qt-pcre     -no-libjpeg     -no-libpng     -no-gif     -no-accessibility     -no-cups     -no-gui     -no-qml-debug     -no-sql-mysql     -no-sql-sqlite     -no-opengl     -no-openssl     -no-xcb     -no-icu     -verbose     -release     -prefix /home/pierre/www/asteroid/build/tmp-glibc/work/x86_64-linux/qtbase-native/5.10.1+gitAUTOINC+6c6ace9d23-r0/recipe-sysroot-native/usr     -hostprefix /home/pierre/www/asteroid/build/tmp-glibc/work/x86_64-linux/qtbase-native/5.10.1+gitAUTOINC+6c6ace9d23-r0/recipe-sysroot-native/usr     -bindir /home/pierre/www/asteroid/build/tmp-glibc/work/x86_64-linux/qtbase-native/5.10.1+gitAUTOINC+6c6ace9d23-r0/recipe-sysroot-native/usr/bin     -hostbindir /home/pierre/www/asteroid/build/tmp-glibc/work/x86_64-linux/qtbase-native/5.10.1+gitAUTOINC+6c6ace9d23-r0/recipe-sysroot-native/usr/bin     -libdir /home/pierre/www/asteroid/build/tmp-glibc/work/x86_64-linux/qtbase-native/5.10.1+gitAUTOINC+6c6ace9d23-r0/recipe-sysroot-native/usr/lib     -hostlibdir /home/pierre/www/asteroid/build/tmp-glibc/work/x86_64-linux/qtbase-native/5.10.1+gitAUTOINC+6c6ace9d23-r0/recipe-sysroot-native/usr/lib     -headerdir /home/pierre/www/asteroid/build/tmp-glibc/work/x86_64-linux/qtbase-native/5.10.1+gitAUTOINC+6c6ace9d23-r0/recipe-sysroot-native/usr/include     -archdatadir /home/pierre/www/asteroid/build/tmp-glibc/work/x86_64-linux/qtbase-native/5.10.1+gitAUTOINC+6c6ace9d23-r0/recipe-sysroot-native/usr/lib     -datadir /home/pierre/www/asteroid/build/tmp-glibc/work/x86_64-linux/qtbase-native/5.10.1+gitAUTOINC+6c6ace9d23-r0/recipe-sysroot-native/usr/share     -hostdatadir /home/pierre/www/asteroid/build/tmp-glibc/work/x86_64-linux/qtbase-native/5.10.1+gitAUTOINC+6c6ace9d23-r0/recipe-sysroot-native/usr/lib     -docdir /home/pierre/www/asteroid/build/tmp-glibc/work/x86_64-linux/qtbase-native/5.10.1+gitAUTOINC+6c6ace9d23-r0/recipe-sysroot-native/usr/share/doc     -sysconfdir /home/pierre/www/asteroid/build/tmp-glibc/work/x86_64-linux/qtbase-native/5.10.1+gitAUTOINC+6c6ace9d23-r0/recipe-sysroot-native/etc     -no-glib     -no-iconv     -silent     -nomake examples     -nomake tests     -no-rpath     -no-feature-linkat     -platform linux-oe-g++ 
ERROR: qtbase-native-5.10.1+gitAUTOINC+6c6ace9d23-r0 do_configure: Function failed: do_configure (log file is located at /home/pierre/www/asteroid/build/tmp-glibc/work/x86_64-linux/qtbase-native/5.10.1+gitAUTOINC+6c6ace9d23-r0/temp/log.do_configure.20036)
ERROR: Logfile of failure stored in: /home/pierre/www/asteroid/build/tmp-glibc/work/x86_64-linux/qtbase-native/5.10.1+gitAUTOINC+6c6ace9d23-r0/temp/log.do_configure.20036
Log data follows:
| DEBUG: Executing shell function qmake5_base_preconfigure
| DEBUG: Shell function qmake5_base_preconfigure finished
| DEBUG: Executing shell function do_configure
| Performing shadow build...
| Preparing build tree...
| 
| You need perl in your PATH to make a build from GIT.
| Cannot proceed.
| ERROR: Configuring qt failed. PACKAGECONFIG_CONFARGS was      -sysroot /home/pierre/www/asteroid/build/tmp-glibc/work/x86_64-linux/qtbase-native/5.10.1+gitAUTOINC+6c6ace9d23-r0/recipe-sysroot-native     -no-gcc-sysroot     -system-zlib     -qt-pcre     -no-libjpeg     -no-libpng     -no-gif     -no-accessibility     -no-cups     -no-gui     -no-qml-debug     -no-sql-mysql     -no-sql-sqlite     -no-opengl     -no-openssl     -no-xcb     -no-icu     -verbose     -release     -prefix /home/pierre/www/asteroid/build/tmp-glibc/work/x86_64-linux/qtbase-native/5.10.1+gitAUTOINC+6c6ace9d23-r0/recipe-sysroot-native/usr     -hostprefix /home/pierre/www/asteroid/build/tmp-glibc/work/x86_64-linux/qtbase-native/5.10.1+gitAUTOINC+6c6ace9d23-r0/recipe-sysroot-native/usr     -bindir /home/pierre/www/asteroid/build/tmp-glibc/work/x86_64-linux/qtbase-native/5.10.1+gitAUTOINC+6c6ace9d23-r0/recipe-sysroot-native/usr/bin     -hostbindir /home/pierre/www/asteroid/build/tmp-glibc/work/x86_64-linux/qtbase-native/5.10.1+gitAUTOINC+6c6ace9d23-r0/recipe-sysroot-native/usr/bin     -libdir /home/pierre/www/asteroid/build/tmp-glibc/work/x86_64-linux/qtbase-native/5.10.1+gitAUTOINC+6c6ace9d23-r0/recipe-sysroot-native/usr/lib     -hostlibdir /home/pierre/www/asteroid/build/tmp-glibc/work/x86_64-linux/qtbase-native/5.10.1+gitAUTOINC+6c6ace9d23-r0/recipe-sysroot-native/usr/lib     -headerdir /home/pierre/www/asteroid/build/tmp-glibc/work/x86_64-linux/qtbase-native/5.10.1+gitAUTOINC+6c6ace9d23-r0/recipe-sysroot-native/usr/include     -archdatadir /home/pierre/www/asteroid/build/tmp-glibc/work/x86_64-linux/qtbase-native/5.10.1+gitAUTOINC+6c6ace9d23-r0/recipe-sysroot-native/usr/lib     -datadir /home/pierre/www/asteroid/build/tmp-glibc/work/x86_64-linux/qtbase-native/5.10.1+gitAUTOINC+6c6ace9d23-r0/recipe-sysroot-native/usr/share     -hostdatadir /home/pierre/www/asteroid/build/tmp-glibc/work/x86_64-linux/qtbase-native/5.10.1+gitAUTOINC+6c6ace9d23-r0/recipe-sysroot-native/usr/lib     -docdir /home/pierre/www/asteroid/build/tmp-glibc/work/x86_64-linux/qtbase-native/5.10.1+gitAUTOINC+6c6ace9d23-r0/recipe-sysroot-native/usr/share/doc     -sysconfdir /home/pierre/www/asteroid/build/tmp-glibc/work/x86_64-linux/qtbase-native/5.10.1+gitAUTOINC+6c6ace9d23-r0/recipe-sysroot-native/etc     -no-glib     -no-iconv     -silent     -nomake examples     -nomake tests     -no-rpath     -no-feature-linkat     -platform linux-oe-g++
| WARNING: exit code 1 from a shell command.
| ERROR: Function failed: do_configure (log file is located at /home/pierre/www/asteroid/build/tmp-glibc/work/x86_64-linux/qtbase-native/5.10.1+gitAUTOINC+6c6ace9d23-r0/temp/log.do_configure.20036)
ERROR: Task (/home/pierre/www/asteroid/src/meta-qt5/recipes-qt/qt5/qtbase-native_git.bb:do_configure) failed with exit code '1'
NOTE: Tasks Summary: Attempted 915 tasks of which 904 didn't need to be rerun and 1 failed.

Summary: 1 task failed:
  /home/pierre/www/asteroid/src/meta-qt5/recipes-qt/qt5/qtbase-native_git.bb:do_configure
Summary: There were 2 ERROR messages shown, returning a non-zero exit code.
```

I'm using Debian 9.3 and I have Perl 5.24.1 installed.

I tried to build eef823e05fb235403d754d04955c34d744dc530e instead but I get the same error with qtbase-native 5.9.4.

FlorentRevest, 2018-02-17:
Hi @Rudloff,
Thanks for reporting. The problem seems to be in meta-qt5. Could you please try to replace the meta-qt5 folder in your src/ directory with a git clone of the master branch of https://github.com/meta-qt5/meta-qt5 ?
If the problem persists, maybe you should report the issue to meta-qt5 instead (I believe they use a mailing list for that)

Rudloff, 2018-02-17:
I get the same error when I clone meta-qt5 manually.
I will try to report the issue upstream.

FlorentRevest, 2018-02-17:
Thank you! I'll let this issue open for now, feel free to comment again if there is progress from upstream meta-qt5.

Rudloff, 2018-02-24:
I finally found the problem! For some reason, I had [noe-which](https://github.com/npm/node-which) in my path, so it was used instead of the native `which` and the configure script didn't like it.

closed 2018-02-24 by Rudloff