AsteroidOS/asteroid #56: dory: vibration no longer works

craftyguy, 2018-03-18:
I seem to recall vibration working on dory a few months back when I last tested asteroidos on my watch, but after installing the latest nightly (as of yesterday), it no longer works.

I can confirm this by running `test_vibrator` on the device, and nothing happening.

craftyguy, 2018-03-18:
It seems that it *does* work, it's just disabled by default. I don't know if this is intended or not. If it is, feel free to close this issue but having some note to users that it is disabled by default might be helpful!

FlorentRevest, 2018-03-18:
They should actually be enabled by default. I'll give it a look.
However, test_vibrator is not supposed to work on dory.

FlorentRevest, 2018-03-27:
I suspect this https://github.com/AsteroidOS/meta-asteroid/commit/610307e9cad40b1871d55cf171dae603208a7d75 could have been the source of your issue.
Could you please try the next nightly (when it will be built, in a couple of days) and tell me if your problem is fixed?

craftyguy, 2018-03-27:
Sure! Would it be possible to get this fix with a 'opkg upgrade' on an existing install of asteriodos?

FlorentRevest, 2018-03-27:
That fix should be included in an opkg upgrade indeed.

craftyguy, 2018-04-09:
It seems to work now, thanks!

closed 2018-04-09 by craftyguy