AsteroidOS/asteroid #58: Always on screen for sony smartwatch 3

Doomsdayrs, 2018-04-08:
I always have found it to be a great tool when using the watch, simply looking at it for the time instead of having to turn it on with a button. And i find it would be a great addition to the watch itself. 

That's just an idea i had for the sw3.

FlorentRevest, 2018-04-08:
Thanks for your suggestion @Doomsdayrs, I edited the name of your issue to be more generic (there is no reason why always on display should only work on the SW3).
There is no plan for this as of today because it is a very complex problem that would require a lot of work but I let your issue open to keep track of it.

Doomsdayrs, 2018-04-08:
@FlorentRevest Understood, lots of people can benefit from it if it ever happens 😄

c-e-p-x-u-o, 2018-04-11:
I'm also would like like this feature!

Really when I tried AsteroidOS with dory it was possible to say some command (I don't remember it) on the clock to prevent display turn off. But IPS backlight eats battery so I sell them.

FlorentRevest, 2018-04-12:
You can use:
mcetool -D on
to enable "demo mode" that will keep the screen on with full brightness. But your CPU won't enter deep sleep mode and the battery will be emptied very quickly. An actually efficient "always on screen" mode would require much more work and deeper integration to various system components.

c-e-p-x-u-o, 2018-04-12:
Exactly! But,
a. I'm absolutely sure I was able to set minimal brightness after "mcetool -D on"
b. Are you sure if it's possible to deep sleep and update the screen every second?
c. Fuck fuel economy! The only way I can use the watch is when display is always on and shows actually seconds. (Otherwise it's just a garbage on my wrist.) Dory with not new battery was enough for about 8-12 hours, But as I understand the main load was an IPS white LED emitter, not CPU. I'd like to test some wathc with OLED screen but don't know which one to choose. The enough operating time between battery charge is 24 hours for me.

FlorentRevest, 2018-04-12:
a) Yes you can
b) It is *technically* possible (android wear does it) but very complex in practice. Again, it requires a major architectural overhaul that is not in sight for the near future.
c) I don't think an OLED screen with asteroid in demo mode, without the CPU entering deep sleep (and kernel putting various buses/devices at sleep as well) can get you 24 hours of battery life, but feel free to experiment.

c-e-p-x-u-o, 2018-04-12:
Could you suggest me the watch with OLED?

FlorentRevest, 2018-04-12:
I'm afraid I am clueless about that.

eLtMosen, 2018-04-12:
Lenok (lg watch r) and bass (lg watch urbane) both have P-OLED displays, became cheap and are nicely supported!

Rudloff, 2018-04-16:
Having this (even a rough version) would be useful on watches that don't support vibration (yet) since it would allow to see more easily if there is a new notification.

Doomsdayrs, 2018-04-20:
The SW3 has a tft display if i am not incorrect

violoncelloch, 2018-05-15:
I would also like an always on display mode.
only showing the watchface without the background would be extremely useful for watches with OLED screens

andrew-ledawson, 2018-05-18:
@Doomsdayrs the Smartwatch 3 has a transflective display, which is notable because it doesn't need backlight for greyscale images.  The Smartwatch 3 may have panel self-refresh which would explain how to have good battery life for always-on mode.  If so, the challenge may be figuring out the commands to the display driver IC.

NeonHorizon, 2018-05-31:
Just tried Asteroid on my Zenwatch 3 and I love it, its so smooth and responsive and looks fantastic, great work guys! Apologies for the "me too" post but I too would really like to see an always on, I always thought it was one of Android Wears killer features compared to the competition, would be great to see it on Asteroid.

eLtMosen, 2018-06-01:
I looked into the matter and reflashed a lenok to stock wearOS.
When you do `top` via `adb shell` it is kind of obvious that the process `com.google.android.wearable.ambient` only is doing work when the watch is showing the always-on-screen.
Further search brought me to https://developer.android.com/training/wearables/apps/always-on were the usage of `AmbientModeSupport` is explained. How exactly the low-power ambient mode works is not explained, maybe someone else will pick on from here and can find more sources now that we have a name for the thing.
I have the strong feeling that it is more simple than we imagined.
All the talk is just about oled displays turing off black pixels. 
So use much black and update only every minute and you have your low-power mode on oled? Can't be that simple, can it?
How to find out what com.google.android.wearable.ambient actually does?

kmccurley, 2019-10-29:
As far as I'm concerned this is reason enough to put my watch back in a box. Or just send it to electronic waste. A watch that doesn't tell you the time when you just look at it is a 40 year regression - not exactly what I would call advanced technology.

Doomsdayrs, 2019-10-29:
Indeed, it's sad that there hasn't been much referenced development on this @kmccurley

MagneFire, 2020-05-15:
I've got the display to stay on while letting the soc got to deep-sleep mode!
Currently it's very experimental, but at least there is some progress :)
To get it to work changes are needed for `mce`(https://github.com/MagneFire/mce/commits/master), `lipstick`(https://github.com/MagneFire/lipstick/commits/master) and `qt5-qpa-hwcomposer-plugin`(https://github.com/MagneFire/qt5-qpa-hwcomposer-plugin/commits/master).

I did look into the decompiled code from ClockworkAmbient. But didn't find anything special. Main things that I noticed from the code are:
- AlarmManager is used to update the screen every minute (function used is `setExact`).
- Services `AmbientService` and `AmbientDream`.
- `doOnDreamingStarted()` in `AmbientService` seems the most important function. It does:
  + Calls makeRunningActivitiesEnterAmbientLocked() which calls `IAmbientActivityCallbacks` -> `enterAmbient()`
  + Uses AlarmManager to schedule next wakup (`maybeScheduleNextWakeup`)
  + Makes the system doze (`maybeSendDozeCommand`) (enter deep sleep).

Initially I thought that the `mce` tool was a great starting point to look into, since it is responsible for screen brightness and screen on/off stuff.
`mce` uses a state machine to control the display. This is the main thing to look at(https://github.com/MagneFire/mce/blob/master/modules/display.c#L7559).
At some point it fades the screen to black(STM_WAIT_FADE_TO_BLACK) and turns off the display(mce_fbdev_set_power).

`mce` sends a D-Bus command to `lipstick` (the compositor) (setUpdatesEnabled() function) which sends a command to the Qt5 backend to turn off the display.
The Qt5 backend uses the libhybris backend to control the display(libhybris uses the hwcomposer module for this). the `sleepDisplay` function is used to turn off the display.
All we need to do is change the `HWC_POWER_MODE_OFF` to `HWC_POWER_MODE_DOZE_SUSPEND`(https://android.googlesource.com/platform/hardware/libhardware/+/master/include/hardware/hwcomposer_defs.h#299) to keep the screen awak while letting the system enter deep sleep.

This is all just a very minimal way to get it to work. Current issues are:
- Kernel panic when screen turns off(this doesn't reboot the system though ;)).
- Tap-to-wake is broken :(
- Doesn't update the screen every minute yet, Currently press the power button or use tilt-to-wake feature to update the screen.

Currently I am using the LPM feature of `mce` to lower the brightness before entering deep-sleep. This may or may not be the best solution.

I've tested this for about 4 hours now and the battery drained about 3% with Bluetooth disabled and manually waking the screen every hour.

<details>
  <summary>Relevant part of the kernel panic</summary>
  <p>

```
  May 15 05:55:55 sturgeon kernel: mdss_dsi_off: Panel power off failed
May 15 05:55:55 sturgeon kernel: ------------[ cut here ]------------
May 15 05:55:55 sturgeon kernel: WARNING: at drivers/video/msm/mdss/mdss_mdp_intf_cmd.c:990 mdss_mdp_cmd_stop+0x374/0x444()
May 15 05:55:55 sturgeon kernel: intf 2 unblank error (-598599664)
May 15 05:55:55 sturgeon kernel: CPU: 0 PID: 33 Comm: kworker/u8:1 Tainted: G        W    3.10.40 #1
May 15 05:55:55 sturgeon kernel: Workqueue: autosleep try_to_suspend
May 15 05:55:55 sturgeon kernel: Backtrace:
May 15 05:55:55 sturgeon kernel: [<c09089e8>] (dump_backtrace+0x0/0x104) from [<c0908b7c>] (show_stack+0x18/0x1c)
May 15 05:55:55 sturgeon kernel:  r7:000003de r6:c0bf5f91 r5:00000009 r4:dd5e3b50
May 15 05:55:55 sturgeon kernel: [<c0908b64>] (show_stack+0x0/0x1c) from [<c090bd44>] (dump_stack+0x20/0x28)
May 15 05:55:55 sturgeon kernel: [<c090bd24>] (dump_stack+0x0/0x28) from [<c09096a8>] (warn_slowpath_common+0x50/0x70)
May 15 05:55:55 sturgeon kernel: [<c0909658>] (warn_slowpath_common+0x0/0x70) from [<c0124f20>] (warn_slowpath_fmt+0x38/0x40)
May 15 05:55:55 sturgeon kernel:  r9:00000003 r8:c0f3b1d4 r7:c0ea4230 r6:00000001 r5:dc521810
                                 r4:dc520810
May 15 05:55:55 sturgeon kernel: [<c0124ee8>] (warn_slowpath_fmt+0x0/0x40) from [<c0383280>] (mdss_mdp_cmd_stop+0x374/0x444)
May 15 05:55:55 sturgeon kernel:  r3:00000002 r2:c0bf5f75
May 15 05:55:55 sturgeon kernel: [<c0382f0c>] (mdss_mdp_cmd_stop+0x0/0x444) from [<c036a060>] (mdss_mdp_ctl_stop+0xf0/0x398)
May 15 05:55:55 sturgeon kernel: [<c0369f70>] (mdss_mdp_ctl_stop+0x0/0x398) from [<c038a384>] (mdss_mdp_overlay_off+0xb8/0x334)
May 15 05:55:55 sturgeon kernel: [<c038a2cc>] (mdss_mdp_overlay_off+0x0/0x334) from [<c03a21e8>] (mdss_fb_blank_blank+0x12c/0x1a4)
May 15 05:55:55 sturgeon kernel:  r9:00000002 r8:00000001 r7:dc4e83ac r6:dc4e8344 r5:00000003
                                 r4:dc4e8270
May 15 05:55:55 sturgeon kernel: [<c03a20bc>] (mdss_fb_blank_blank+0x0/0x1a4) from [<c03a23e4>] (mdss_fb_blank_sub.isra.10+0x184/0x280)
May 15 05:55:55 sturgeon kernel:  r8:00000000 r7:dc56f044 r6:00000001 r5:c0ea61f8 r4:dc4e8270
May 15 05:55:55 sturgeon kernel: [<c03a2260>] (mdss_fb_blank_sub.isra.10+0x0/0x280) from [<c03a3988>] (mdss_fb_suspend_sub+0xc0/0x10c)
May 15 05:55:55 sturgeon kernel:  r7:dc56f044 r6:c0fb9764 r5:00000000 r4:dc4e8270
May 15 05:55:55 sturgeon kernel: [<c03a38c8>] (mdss_fb_suspend_sub+0x0/0x10c) from [<c03a3a1c>] (mdss_fb_pm_suspend+0x48/0x5c)
May 15 05:55:55 sturgeon kernel:  r7:dc56f044 r6:c0fb9764 r5:dc56f010 r4:dc4e8270
May 15 05:55:55 sturgeon kernel: [<c03a39d4>] (mdss_fb_pm_suspend+0x0/0x5c) from [<c04634e0>] (platform_pm_suspend+0x34/0x6c)
[... cut ...]
May 15 05:55:55 sturgeon kernel: [<c0144680>] (kthread+0x0/0xb4) from [<c0106558>] (ret_from_fork+0x14/0x3c)
May 15 05:55:55 sturgeon kernel:  r7:00000000 r6:00000000 r5:c0144680 r4:dd483e68
May 15 05:55:55 sturgeon kernel: ---[ end trace 20bfc2e6814626d0 ]---
May 15 05:55:55 sturgeon kernel: mdss_mdp_ctl_stop: error powering off intf ctl=0
May 15 05:55:55 sturgeon kernel: active wakeup source: IDLE_ON_WAKELOCK
May 15 05:55:55 sturgeon kernel: PM: Some devices failed to suspend
```
</p>
</details>

Doomsdayrs, 2020-05-15:
After 2 years, we have a genius! Thank you @MagneFire

MagneFire, 2020-05-26:
So I have got everything mostly working.

There is still a small bug where the time is one minute behind in ambient mode. This happens because the watch goes to sleep before the watchface has completed painting the Canvas'. 

I propose the following to solve the issue:
- mainscreen: Start timer when  `onDisplayAmbientUpdate` event from `Lipstick.compositor`
- watchface: Handles `onDisplayAmbientUpdate` event from `Lipstick.compositor`
- watchface: Paints Canvas
- watchface: On Canvas `onPainted` -> signal mainscreen somehow that it is done painting.
- mainscreen: Move grid
- mainscreen: Enter deep sleep.

And if the watchface does not obay the new API we get:
- mainscreen: Start timer when  `onDisplayAmbientUpdate` event from `Lipstick.compositor`
- watchface: Ignores `onDisplayAmbientUpdate` event from `Lipstick.compositor`
- mainscreen: Timeout of `onDisplayAmbientUpdate` timer.
- mainscreen: Move grid.
- mainscreen: Enter deep sleep.

Finally, here is some relevant code that watchfaces can use:
```qml
Connections {
    target: Lipstick.compositor
    onDisplayAmbientEntered:
    onDisplayAmbientLeft:
    onDisplayAmbientUpdate:
}
Lipstick.compositor.displayAmbient // Is the display currently in ambient mode?
Lipstick.compositor.ambientEnabled
``` 

I have adjusted the following components:
- [MCE](https://github.com/MagneFire/mce/commits/master) (WIP): Controls whether ambient mode should be enabled.
- [lipstick](https://github.com/MagneFire/lipstick/commits/master) (WIP):
- [asteroid-launcher](https://github.com/MagneFire/asteroid-launcher/commits/ambientmode) (WIP): Update time and move grid, to prevent burn in.
- [timed](https://github.com/MagneFire/timed/commits/master) (WIP): 
- [qt5-qpa-hwcomposer](https://github.com/MagneFire/qt5-qpa-hwcomposer-plugin/commits/master) (Complete): Set display to ambient mode.

Here is a simple diagram to show what I think the ideal flow should be.
[image]

Another small issue has something to do with timed.
When I schedule a alarm event. A dbus signal is send to `com.nokia.voland` when the alarm fires. Which always starts the `alarm-presenter` from `asteroid-alarmclock`. Do you know if there is a proper way to filter `clock` events for `asteroid-alarmclock` with systemd? This would remove the need for any changes in `timed`.

Here is a short demo of it all in action:
[image]

532910, 2020-05-26:
Is it usable with one second update (with second hand)?

MagneFire, 2020-05-26:
Currently I have about 40~50% battery left at the end of the day. This is with a one minute update.

Of course it is be possible to change the update to something arbitrary, this will have a significant hit on the battery life though. I think your watch won't make it through the day then...

One thing you can do is disable tilt-to-wake, this will increase the battery life.

eLtMosen, 2020-05-26:
Great work, fantastic!
I think the burn-in offset between the minutely watchface placements needs to be even bigger.
With heavy weight fonts i would guess it is necessary to offset by the width of the thickest element on screen to prevent the center most pixels of those elements/fonts to be always on?

Since i am slow, could you link to the stock watchface you ambienized already so i can see what is happening and learn from the example?

MagneFire, 2020-05-26:
Thanks! :)

I agree on the changes for burn in. I am actually thinking that we may also want to desynchronize the watchface and wallpaper. And maybe add a dark filter for the watchface.

Ambient mode settings will then become something like:
- Ambient brightness.
- Desync watchface and wallpaper.
- Darken colors in ambient mode.

I didn't ambienize any watchface. The demo shows behaviour without any change to the watchfaces. The idea is something I am thinking of implementing.
What do you think of this idea? If it isn't really clear I can create a simple prototype for this.

eLtMosen, 2020-05-26:
Understood! Much easier than i thought.
My 2 cents
- Ambient mode should be very readability optimized since in most situations the watch will be in some distance to your eyes when you peek at it.
- Hide Wallpaper should be stock setting to visually differentiate between AmbientMode / active mode so casual user do not mix them up. For example the user might expect there is input possible already but needs to do an action first to end AmbientMode. Some users might not recognize the Ambientmode at all and question if there is any if it is not visually differentiated. Enable Wallpaper only for advanced users from settings.
- It might be a cool effect if the wallpaper fades-in behind the watchface when Ambientmode ends.

MagneFire, 2020-05-27:
Thanks for your feedback!

I put together a quick demo of your feedback. What do you think of it?

[image]

kmccurley, 2020-05-27:
There is always a balance between having too many options in the settings and providing adequate usability for every use case. I stopped using Asteroid because it wouldn't keep the time showing, but some people might think battery life is more important. This suggests to me that it should be controlled by at least one setting. Other settings like brightness in ambient mode might be deferred if there is a disagreement, but I think having some kind of ambient mode with the time displayed is a showstopper and I'm a big fan of this demo.

MagneFire, 2020-05-27:
@kmccurley I guess the default will be what you see in the short demo. With the only setting you have to change is enabling ambient mode.

Ambient mode, obviously, consumes more batter power. For this reason alone, the default might be to have ambient mode disabled by default. Unless @FlorentRevest finds this a super important feature that should be enabled by default :wink: 

All other options are considered `advanced`.

eLtMosen, 2020-05-27:
Gorgeous @MagneFire !
I second @kmccurley when thinking about special user demands. +1 For the requested settings option to disable the AmbientMode.
To overthink a little, i could see users want to only deactivate the AmbientMode only by certain actions. E.g. some users complained about "Too much wallpaper/color" and demanded a black wallpaper to be stock. (Which Kido did not grant to my relief)
Setting:
- End AmbientMode with -> checkbox options: tilt-wake, long tap, button press 

Philosophy wise, i would advise to keep the AmbientMode on for stock delivery
- People do not need to be educated that asterid now has one but see it right after flash
- It is a great achievement (of yours) that earned to be advertised properly and the community can be proud of.
- Competitors all have AmbientMode as regular feature, people not familiar to asteroidOS expect it anyway to be stock and missed it in the past when migrating.

FlorentRevest, 2020-05-28:
I am all-in for having AmbientMode enabled by default. This is what most users want, this is what should be set as default.

There should be an option to disable AmbientMode and come back to what we have right now (screen off) in asteroid-settings because it's still a setup that can make sense in some situations. However, I think that anything else should be either not-configurable (have sane default) or only configurable from the command line (with mcetool for instance). For example, the brightness of ambient mode or whether we should show a wallpaper in ambient mode or not. By the way I agree with eLtMosen that there should be no wallpaper in ambientmode by default, it saves some battery life, improves the readability and probably makes the screen last longer too.

I will review the patches when my time will permit but I of course share the excitement of everyone here about this feature which we have been missing for a long time. :)

closed 2020-08-13 by FlorentRevest