AsteroidOS/asteroid #61: RTL Interface glitches

yarons, 2018-04-18:
This was done for Hebrew, I can take a look at other RTL languages and see what's going on over there, regardless, all the directionality issues I'm reporting are relevant to both Persian/Farsi, Kurdish, Hebrew, Arabic, Urdu, N'Ko, Divehi, Punjabi and several others writing the opposite direction.

Both the Farsi and Kurdi glyphs have their edges overlapping causing a brighter color on the overlapping parts:
[image]
  
At the Units screen, the buttons appear in English (1), maybe a colored toggle will eliminate the need for translation (Green/Red), also as the semicolon suggests the button should be in the opposite direction (2):
[image]

Settings screen, the icons should be on the right, not on the left as they appear now:
[image]

Some weird Unicode boxes appears all over the place, not sure what to do with it:
[image]

At the Events screen there are several issues, a partial Unicode box (1), plural appears as singular (2 - it shows: 1 Events), the sliding direction and the appearance of the days circles should slide to the opposite direction (3) - from right to left instead of left to right.
[image]

If needed I can break these to individual bugs and add my insights about Farsi and Kurdish but this is all for now, thanks!

FlorentRevest, 2018-04-18:
Thanks a lot for your insights Yaron! I would never have thought of all those things by myself... This is really helpful! :) 
One issue for everything related to RTL languages is enough.
I don't think I will get those things fixed for Asteroid 1.0 but this is definitely going to be an important fix after 1.0 is released.

yarons, 2018-04-18:
I'll check that again once 1.1 will be released, thanks!

sqwerty13, 2018-05-04:
The first issue's reason is easy but it's fix way may be very hard :(

It occur because Persian (or Arabic or...) alphabet characters have often 4 mode (head join, last join, both join and not joined to other chars) see it:

https://en.m.wikipedia.org/wiki/Persian_alphabet

For characters that are in both side join, they's glyph a little goes together to make sure that it seems nice.

مـــــــــثــــلـا ایــــــنـــــجـــــــوری

"ـــــجـــــ" 

So, when we set a textblock's opacity half, that excrescences (half opacity) puts together and 50%+50% = 100%! 

(I'm very sorry for bad explain 😅)

yarons, 2018-05-04:
@sqwerty13 
Yeah, I'm aware, I can speak some Arabic as well 😄

The problem is the algorithm that draws the glyphs, I'll explain it the way I was explained once and then we'll see if it's worthwhile digging into it.

If we drop the opacity to 50 percent for each glyph individually then you're right but what if we simply draw the entire sentence and then apply transparency to the whole unit?
If I want to highlight an Arabic text and add some border to it the same things happens, so as far as I can remember the first algorithm draws the text and the second adds the border to the final form, this way the borders are not per glyph but per whole segments (Unless you have non left joined characters where there's no such problem at least on the left side).

I can ask some friends of mine and maybe even try and get some famous RTL engineers to share their solution.

Thanks!

FlorentRevest, 2018-05-04:
@yarons, fair enough. maybe we could fix this problem by wrapping the current Label https://github.com/AsteroidOS/qml-asteroid/blob/master/src/controls/qml/SpinnerDelegate.qml#L37 into an Item.