AsteroidOS/asteroid #63: BUG: Unable to continue from select language Zenwatch

MegaArthur, 2018-05-08:
I have just flashed AsteroidOS on my Asus Zenwatch 1 and I am having a pretty big issue, that is, the watch displays the select language screen, but nothing can be selected.

I can scroll through the list of languages without a problem, but the select button-arrow doesn't do anything. I've tried tapping everywhere and doing stuff with the watch button, but nothing works.

I have been able to ssh into the watch, so maybe it could be possible to set some kind of property so the language selection screen doesn't pop up?

FlorentRevest, 2018-05-09:
Hi MegaArthur, this is unfortunate! I don't have an anthias to try it out but I would like to know if this could be a generic touch screen problem or something wrong in the first run dialog that could potentially affect other watches.
Could you share logs from journalctl --no-pager user@1000 ? 
You can skip the first run dialog by setting the dconf value /org/asteroidos/launcher/firstRunDone to true.

MegaArthur, 2018-05-14:
@FlorentRevest Is there any way I can help? For example test bugfixes and provide logs?

FlorentRevest, 2018-05-14:
Unfortunately, I don't notice anything wrong in the logs you uploaded.
The "touchscreen not working after sleep" issue has been confirmed by other users but this "confirmation buttons not working" surprisingly hasn't  been encountered by anyone else. Could you just confirm if it still happens with this image: https://release.asteroidos.org/1.0/anthias/ ?

MegaArthur, 2018-05-14:
The new release doesn't seem to have fixed it, and the buttons still don't work, same happens for the calculator app, some digits can be pressed, and some cannot.

MegaArthur, 2018-06-06:
@FlorentRevest
Any progress in diagnosing the issue?

FlorentRevest, 2018-06-07:
Hi @MegaArthur, unfortunately I don't have the watch to investigate that issue (it only affects anthias) and given its complexity, I'm afraid I won't be able to solve it without the watch.

timttmy, 2018-07-06:
I have a spare Zenwatch 1 I would like to donate to the project to get this bug fixed. @FlorentRevest would you accept the donation? If so please tell me where to send it. :)

FlorentRevest, 2018-07-07:
Hey @timttmy! Sure, that could work.This is very kind of you.
I can't promise I will be able to fix the bug but I can definitely give it a closer look if I have the watch. Feel free to drop me an email or an IRC private message and I'll reply with my address.

FlorentRevest, 2018-07-20:
Regarding the "touch after screen off doesn't work" part of this issue, please refer to: https://github.com/AsteroidOS/meta-anthias-hybris/issues/1#issuecomment-406745180

MegaArthur, 2018-07-26:
I think the second part of the issue might be a hardware problem with my watch, so I'll close the issue, but also suggest one thing: maybe a switch in the settings or dconf that could change the position of buttons slightly? I know I'm just asking for myself, but, if at any point you'd have nothing to do, then maybe this?

closed 2018-07-26 by MegaArthur

FlorentRevest, 2018-07-26:
We can't take into account every hardware problems ever. If you want to change the interface you can edit the QML and recompile Asteroid ;)