AsteroidOS/asteroid #64: Unable to reboot

mikey242, 2018-05-16:
I'm on an LG Urbane and whenever the OS crashes and reboots/switches off I see the LG logo briefly then it goes off and nothing happens. Manually trying to switch the watch on again causes the same thing to happen. Initially I was able to get it to boot again by using the fastboot command ``fastboot continue``, however that no longer works and reinstalling doesn't seem to help either.

The watch usually crashes when I am changing things in the settings, most often the watchface and wallpaper.

Thanks again for this OS, really happy to have an alternative.

mikey242, 2018-05-16:
OK managed to get it to boot again (after reinstalling). Went through the tutorial and immediately after when into settings. As soon as the settings screen appeared the watch switched off and was unable to boot again.

coderobe, 2018-05-16:
I'm also on an LG G Watch Urbane (bass) and i can't reproduce this

mikey242, 2018-05-16:
Yeah I haven't seen anyone reporting it with an LG Urbane, wonder if I have a hardware fault. Is there a log somewhere I could produce?

FlorentRevest, 2018-05-16:
Also, when the watch doesn't reboot, could you try https://github.com/AsteroidOS/asteroid/issues/66#issuecomment-389622262 ?

mikey242, 2018-05-16:
I'm going to hazard a guess that this problem is related to my crappy aftermarket usb dock I have. It is very sensitive and does not easily maintain a steady connection. I am struggling to get it to connect to SSH, I succeeded once but didn't manage to get the info I needed. If I move the watch about on its base this can trigger a reboot.

FlorentRevest, 2018-05-16:
Mhhh... unfortunately there is nothing of interest in those logs indeed.
The problem seems more serious than a "crappy usb dock". You would have had the same problem on Android Wear if it was the usb dock.
There must be something wrong on Asteroid's side but I am completely clueless about that one :|

mikey242, 2018-05-16:
Indeed this problem never happened with Android, but I suspect that the reboots happened as a result of the connection with the charging dock (after I installed it yesterday I left it on the dock while using it as the battery was low). I can fairly consistently cause it to reboot now.

eLtMosen, 2022-08-06:
This sounds suspiciously like the problems i had with lenok and bass before swapping in new batteries.

eLtMosen, 2025-04-24:
Sorted as low battery capacity

closed 2025-04-24 by eLtMosen