AsteroidOS/asteroid #65: Misfit Vapor support

denysvitali, 2018-05-16:
As per title, the Misfit Vapor smartwatch [doesn't seem to be supported](https://asteroidos.org/wiki/porting-status/). But the reason isn't really clear to me: "Impossible ports because of lack of pinouts"  
  
Can you elaborate on this? The Vapor should have an USB connection, which I guess should be enough. I still haven't figured it out how to put it in fastboot mode, but that seems to be the only problem.  
There might also be a closed-source kernel problem, but I don't think the Vapor's CPU is rather different from other smartwatches

FlorentRevest, 2018-05-16:
Thanks a lot for your report.
I have classified watches based on quick google searches. I don't own any of those smartwatches so it's hard to see if they have a pinout or not. After a double check, it looks like I made a mistake indeed. I'll update the website according to your comment.

Regarding the kernel. Android Wear usually does a good job at publishing kernel source tree somewhere on https://android.googlesource.com/ but that usually requires to know the codename of your watch (which you can probably find if you get an adb access on it)

denysvitali, 2018-05-16:
Unfortunately `mullet` doesn't seem to be available, and [Misfit doesn't provide any open source support for the Vapor](https://twitter.com/DenysVitali/status/979627469983580160?s=19). If you want you can re-open the issue as I will probably try a port :+1:

FlorentRevest, 2018-05-16:
Err, good luck with that... Maybe you should try to contact someone else. The community manager clearly has no clue what the kernel or GPL are

deadman96385, 2018-09-30:
Mullet is available 

https://android.googlesource.com/kernel/msm/+/android-msm-mullet-3.18-nougat-dr1-wear
https://android.googlesource.com/kernel/msm/+/android-msm-mullet-3.18-oreo-wear-dr

denysvitali, 2018-09-30:
Thank you! I'll try to do the port as soon as I have more free time :+1:

Icarodesign, 2019-08-30:
Hi, i got a vapor and i want install asteroid, you resolved the issues. i cant install.
Thanks you

jrtberlin, 2020-04-14:
@denysvitali did you had any success? I have Misfit Vapor and have tried to boot the mullet kernel built from source with no success (depending on the GCC version it did not boot or just booted the installed kernel). I thought I could extract a prebuilt one from a software update to build TWRP, so that I could backup the original ROM in case I need it. I dumped an OTA update, but that did not include a boot.img and most relevant partitions we're only present as a diff.

denysvitali, 2020-04-14:
Hello @jrtberlin ,

unfortunately I haven't looked into the issue for a while now. I'll make sure to look into it again and hopefully we'll be able to extract the boot.img 🎉

FlorentRevest, 2020-04-14:
Oh, I see you’re based in Zürich @denysvitali. If you want to give porting AsteroidOS a try, I’d be happy to meet for a beer and help out. :)

denysvitali, 2020-04-14:
Sure thing, why not? I guess that for now we'll have to posticipate it after the lockdown 😅

I'll try my luck with getting the factory image out of my device / compiling a kernel on my own and see how far I can get for now

jknigga, 2023-01-31:
@denysvitali - I know it's been a few years but did you have any luck with this?

Icarodesign, 2023-01-31:
Hey, yea was a couple years a go, but the answer is no, many things don’t
work with the os, after try I can’t install again the original os. And I
send the watch to guarantee and they send me a new watch. (New model) jeje.
Sorry, I don’t know if the update can make all be functional, but nothing
to lost, try and good luck.