AsteroidOS/asteroid #66: Unable to boot in LG G Watch

R4ULSP, 2018-05-16:
Hello, I installed the ROM in my G Watch, after flashing the files and reboot the watch to boot into AsteroidOS the image gets freezed in the boot animation. Then I tried to reinstall after a full wipe of the watch, this time it didn´t started the boot animation.

FlorentRevest, 2018-05-16:
This could be a duplicate of https://github.com/AsteroidOS/asteroid/issues/66
If you feel comfortable enough, could you try this out https://asteroidos.org/wiki/boot-process/#troubleshooting ? The instructions are written for a temporary install. If you have a real install, you'll need to use:
```
# . /machine.conf
# mkdir /rfs
# mount /dev/$sdcard_partition /rfs

# sed -i '/volatile/d' /rfs/etc/fstab
# mkdir /rfs/var/log/journal

# chroot /rfs /bin/journalctl
```
instead

FlorentRevest, 2018-05-16:
It can theoretically be done from Windows, it only requires an adb access. But without some knowledge of Linux, this could be tedious.

FlorentRevest, 2018-05-19:
This could be a duplicate of https://github.com/AsteroidOS/asteroid/issues/68
It looks like some version of fastboot on windows don't flash asteroid correctly. Could you try different versions of fastboot? It has been reported that some windows users have managed to flash asteroid correctly but at that point we don't know which version of fastboot they used.

adamprice56, 2018-05-20:
I had the same issue, It worked perfectly after booting into Tails 3.7 and installing using the latest adb and fastboot.  

I was running adb (1.0.32) and fastboot  revision eac51f2bb6a8-android on Windows 10

peepuk, 2018-06-02:
Did have an issue after the first flash (the watch was in a reboot loop). Just flashed again; the watch booted the second time fine.

(Windows 10, adb 1.0.39, fastboot 0.0.1-4500957, LG Watch Urbane (bass))

frankgerhardt, 2019-10-09:
I have a reboot loop as well. How did you manage to flash it again? How did you get out of the reboot loop? See #96.

jrtberlin, 2019-10-19:
> I have a reboot loop as well. How did you manage to flash it again? How did you get out of the reboot loop? See #96.

On the G Watch (dory) you can enter fastboot if you swipe from top left to bottom right directly after the reset (similar gestures and/or button/cable combinations are most likely present on different models) during the boot splash screen. 
Depending on how far it boots during the process you might also be able to use `adb reboot bootloader` in combination with the `wait-for-device` option. 
e.g.: `adb wait-for-device && adb reboot bootloader`

beroset, 2020-02-14:
If it helps, under Linux (Fedora 31) I just rebuilt and reflashed Asteroid (the latest commit: 61183fdbd8d28c882f19f927a08e40a98295024a ) into my LG G watch without any problems.

comradekingu, 2022-02-04:
@R4ULSP I got this when I installed on an LG G Watch R (lenok) (without setting up AndroidSync if that makes a difference).
Held the button for a while to restart, then tapped the top left and bottom right of the screen (once the LG logo is shown) to get into bootloader mode. :)