AsteroidOS/asteroid #77: Problem building image

wissmarstanley, 2018-08-28:
Hi,

I am new to programming in linux environment and trying to build an image (as described in https://asteroidos.org/wiki/building-asteroidos/ ) but get the following error message

ERROR: Nothing PROVIDES 'libhybris' (but /home/user/asteroid/src/meta-asteroid/recipes-nemomobile/sensorfw/sensorfw_git.bb DEPENDS on or otherwise requires it)
libhybris was skipped: PREFERRED_PROVIDER_virtual/mesa set to mesa, not libhybris
NOTE: Runtime target 'sensorfw' is unbuildable, removing...
Missing or unbuildable dependency chain was: ['sensorfw', 'libhybris']
ERROR: Required build target 'asteroid-image' has no buildable providers.
Missing or unbuildable dependency chain was: ['asteroid-image', 'sensorfw', 'libhybris']

What is wrong? I have installed the libhybris in the Asteroid folder but still obtain this error message.

FlorentRevest, 2018-08-28:
Hi @wissmarstanley,
I'm glad you're interested in AsteroidOS ;) This issue has been fixed a couple of days ago. Could you try to wipe your entire asteroid build tree and compile again? It should work with the latest layers.

wissmarstanley, 2018-09-01:
Hi,
Thanks for your prompt reply.

I have reinstalled and rebuilt the asteroid tree again as recommended. This time it did manage to create the  zImage-dtb-dory.fastboot file (among others) but not the asteroid-image-dory.ext4 file. 
It gave several warnings (see attached log file)
[20180831180927.log](https://github.com/AsteroidOS/asteroid/files/2342306/20180831180927.log)

Please advice how to solve this.

FlorentRevest, 2018-09-01:
Could you please share the content of */home/stan/asteroid/build/tmp-glibc/work/armv7vehf-neon-oe-linux-gnueabi/geoclue/0.12.99-r0/temp/log.do_install.1916* ?

wissmarstanley, 2018-09-02:
Many thanks, now it seems to work (generating the .ext4 file as well). 

I want to try to develop code for the heart rate on my LG G Watch R, which obviously has the lenok version installed (instead of dory) of Asteroid. How would you recommend to start off doing this? (I am very much new to this platform but experienced in other software languages and application, as a background)

FlorentRevest, 2018-09-02:
I have never investigated heart rate sensors on AsteroidOS so I'm afraid I can not help much with this. You can either approach this using the Android HAL (via libhybris) or by talking directly to the kernel driver but either way it will most likely not be an easy task.
Anyway, this is not related to that issue. You can ask questions about this topic on IRC if you wait long enough for someone to be available.

closed 2018-09-02 by FlorentRevest