AsteroidOS/asteroid #79: Screen unresponsive on Harmony watch

BT1987, 2018-11-16:
Hello,

I have a Kingwear KW88 Pro watch. I'm trying to flash Asteroid OS on the watch using the nightly build from [here](https://release.asteroidos.org/nightlies/harmony) (using SP Flash Tool, fastboot is not working). When I reach the language selection, the screen is not responsive. I can't swipe through the list or confirm the language. It sounds a bit like issue #63. Is there a fix for this? I'm willing to beta test if someone can build Asteroid OS updates for this watch.

Thanks in advance.

FlorentRevest, 2018-11-16:
Mmhhh, I haven't experienced that issue. As long as your nightly has this bbappend https://github.com/AsteroidOS/meta-mtk6580-hybris/blob/master/recipes-qt/qt5/qtbase_%25.bbappend it should be ok... I don't have enough time to try it out myself in the coming weeks

BT1987, 2018-11-16:
Thanks for your answer. I have tried something more and it seems like the screen does respond when you touch it with two fingers.

How can I check if the nightly has the corrent bbappend? Remember I just downloaded it from your release repository. Is there anything else I can try?

Mercu7io, 2019-01-15:
Hello!
How do you do to flash using sp flash?
You direct flash the ext4 to userdata?

Thanks for help

FlorentRevest, 2019-01-26:
@BT1987 I indeed had a build of AsteroidOS flashed on my watch where touchscreen wasn't working. However, I tested a new build (with recent updates of AsteroidOS like the OpenEmbedded upgrade to thud) and touchscreen works just fine on my KW88 now. Since I can't reproduce the issue, I will close it for now. Feel free to re-open it if you still can't use the touchscreen in the next nightly.

@Mercu7io This will hopefully be documented shortly but as of now I am not satisfied with the current flashing tools/method. The ext4 should be flashed on userdata and the zImage to boot.

closed 2019-01-26 by FlorentRevest

Mercu7io, 2019-01-29:
Hello Florent,
Thanks for your reply.
I flash it successfully.
I'm looking for the next nightly with touchscreen working 😄 
Thanks a lot for your work

HeofHIShirts, 2019-10-18:
I just purchased a KW88 Pro, and I can confirm this bug exists. After successfully flashing AsteroidOS with the SP Flashing Tool, I found I could only get the touch face of the watch to respond if I was touching the top of the watch face with another finger. When the face is responsive, things work fairly well, but it's remarkably awkward to have to two-finger touch the watch face to get it to do anything. This was using the harmony nightly from the AsteroidOS page and having to use a compatible official firmware scatter image to get the correct ranges established for flashing using the official flash tool.