AsteroidOS/asteroid #81: Errors while building for switft

Trooner, 2019-01-04:
I am interested in bringing bluetooth to the "swift" watch. Therefore I tried to built according to your build instructions on a standard debian stable system. 

I get the following output:

>>>
Initialising tasks: 100% |###########################################################################################################################################################################| Time: 0:00:07
NOTE: Executing SetScene Tasks
NOTE: Executing RunQueue Tasks
ERROR: qt5-qpa-hwcomposer-plugin-5.6.1+gitrAUTOINC+a85d351855-r0 do_compile: oe_runmake failed
ERROR: qt5-qpa-hwcomposer-plugin-5.6.1+gitrAUTOINC+a85d351855-r0 do_compile: Function failed: do_compile (log file is located at /srv/repos/projekte/smartwatch/asteroid/build/tmp-glibc/work/swift-oe-linux-gnueabi/qt5-qpa-hwcomposer-plugin/5.6.1+gitrAUTOINC+a85d351855-r0/temp/log.do_compile.12708)
ERROR: Logfile of failure stored in: /srv/repos/projekte/smartwatch/asteroid/build/tmp-glibc/work/swift-oe-linux-gnueabi/qt5-qpa-hwcomposer-plugin/5.6.1+gitrAUTOINC+a85d351855-r0/temp/log.do_compile.12708
Log data follows:
| DEBUG: Executing shell function do_compile
| NOTE: make -j 8 MAKEFLAGS=-j 8 OE_QMAKE_COMPILER=arm-oe-linux-gnueabi-gcc  -march=armv7ve -marm -mfpu=neon -mfloat-abi=hard --sysroot=/srv/repos/projekte/smartwatch/asteroid/build/tmp-glibc/work/swift-oe-linux-gnueabi/qt5-qpa-hwcomposer-plugin/5.6.1+gitrAUTOINC+a85d351855-r0/recipe-sysroot OE_QMAKE_CC=arm-oe-linux-gnueabi-gcc  -march=armv7ve -marm -mfpu=neon -mfloat-abi=hard --sysroot=/srv/repos/projekte/smartwatch/asteroid/build/tmp-glibc/work/swift-oe-linux-gnueabi/qt5-qpa-hwcomposer-plugin/5.6.1+gitrAUTOINC+a85d351855-r0/recipe-sysroot OE_QMAKE_CXX=arm-oe-linux-gnueabi-g++  -march=armv7ve -marm -mfpu=neon -mfloat-abi=hard --sysroot=/srv/repos/projekte/smartwatch/asteroid/build/tmp-glibc/work/swift-oe-linux-gnueabi/qt5-qpa-hwcomposer-plugin/5.6.1+gitrAUTOINC+a85d351855-r0/recipe-sysroot OE_QMAKE_CFLAGS= -O2 -pipe -g -feliminate-unused-debug-types -fdebug-prefix-map=/srv/repos/projekte/smartwatch/asteroid/build/tmp-glibc/work/swift-oe-linux-gnueabi/qt5-qpa-hwcomposer-plugin/5.6.1+gitrAUTOINC+a85d351855-r0=/usr/src/debug/qt5-qpa-hwcomposer-plugin/5.6.1+gitrAUTOINC+a85d351855-r0 -fdebug-prefix-map=/srv/repos/projekte/smartwatch/asteroid/build/tmp-glibc/work/swift-oe-linux-gnueabi/qt5-qpa-hwcomposer-plugin/5.6.1+gitrAUTOINC+a85d351855-r0/recipe-sysroot= -fdebug-prefix-map=/srv/repos/projekte/smartwatch/asteroid/build/tmp-glibc/work/swift-oe-linux-gnueabi/qt5-qpa-hwcomposer-plugin/5.6.1+gitrAUTOINC+a85d351855-r0/recipe-sysroot-native=  OE_QMAKE_CXXFLAGS= -O2 -pipe -g -feliminate-unused-debug-types -fdebug-prefix-map=/srv/repos/projekte/smartwatch/asteroid/build/tmp-glibc/work/swift-oe-linux-gnueabi/qt5-qpa-hwcomposer-plugin/5.6.1+gitrAUTOINC+a85d351855-r0=/usr/src/debug/qt5-qpa-hwcomposer-plugin/5.6.1+gitrAUTOINC+a85d351855-r0 -fdebug-prefix-map=/srv/repos/projekte/smartwatch/asteroid/build/tmp-glibc/work/swift-oe-linux-gnueabi/qt5-qpa-hwcomposer-plugin/5.6.1+gitrAUTOINC+a85d351855-r0/recipe-sysroot= -fdebug-prefix-map=/srv/repos/projekte/smartwatch/asteroid/build/tmp-glibc/work/swift-oe-linux-gnueabi/qt5-qpa-hwcomposer-plugin/5.6.1+gitrAUTOINC+a85d351855-r0/recipe-sysroot-native=  -fvisibility-inlines-hidden OE_QMAKE_LINK=arm-oe-linux-gnueabi-g++  -march=armv7ve -marm -mfpu=neon -mfloat-abi=hard --sysroot=/srv/repos/projekte/smartwatch/asteroid/build/tmp-glibc/work/swift-oe-linux-gnueabi/qt5-qpa-hwcomposer-plugin/5.6.1+gitrAUTOINC+a85d351855-r0/recipe-sysroot OE_QMAKE_LDFLAGS=-Wl,-O1 -Wl,--hash-style=gnu -Wl,--as-needed OE_QMAKE_AR=arm-oe-linux-gnueabi-ar OE_QMAKE_STRIP=echo OE_QMAKE_INCDIR_QT=/srv/repos/projekte/smartwatch/asteroid/build/tmp-glibc/work/swift-oe-linux-gnueabi/qt5-qpa-hwcomposer-plugin/5.6.1+gitrAUTOINC+a85d351855-r0/recipe-sysroot//usr/include
| compiling /srv/repos/projekte/smartwatch/asteroid/build/tmp-glibc/work/swift-oe-linux-gnueabi/qt5-qpa-hwcomposer-plugin/5.6.1+gitrAUTOINC+a85d351855-r0/git/hwcomposer/main.cpp
| compiling /srv/repos/projekte/smartwatch/asteroid/build/tmp-glibc/work/swift-oe-linux-gnueabi/qt5-qpa-hwcomposer-plugin/5.6.1+gitrAUTOINC+a85d351855-r0/git/hwcomposer/hwcomposer_backend_v11.cpp
| /srv/repos/projekte/smartwatch/asteroid/build/tmp-glibc/work/swift-oe-linux-gnueabi/qt5-qpa-hwcomposer-plugin/5.6.1+gitrAUTOINC+a85d351855-r0/git/hwcomposer/hwcomposer_backend_v11.cpp: In member function 'void HwComposerBackend_v11::handleVSyncEvent()':
| /srv/repos/projekte/smartwatch/asteroid/build/tmp-glibc/work/swift-oe-linux-gnueabi/qt5-qpa-hwcomposer-plugin/5.6.1+gitrAUTOINC+a85d351855-r0/git/hwcomposer/hwcomposer_backend_v11.cpp:412:13: error: 'class QWindowPrivate' has no member named 'deliverUpdateRequest'
|          wp->deliverUpdateRequest();
|              ^~~~~~~~~~~~~~~~~~~~
| Makefile:475: recipe for target '.obj/hwcomposer_backend_v11.o' failed
| make: *** [.obj/hwcomposer_backend_v11.o] Error 1
| make: *** Waiting for unfinished jobs....
| ERROR: oe_runmake failed
| WARNING: exit code 1 from a shell command.
| ERROR: Function failed: do_compile (log file is located at /srv/repos/projekte/smartwatch/asteroid/build/tmp-glibc/work/swift-oe-linux-gnueabi/qt5-qpa-hwcomposer-plugin/5.6.1+gitrAUTOINC+a85d351855-r0/temp/log.do_compile.12708)
ERROR: Task (/srv/repos/projekte/smartwatch/asteroid/src/meta-swift-hybris/recipes-qt/qt5/qt5-qpa-hwcomposer-plugin_git.bb:do_compile) failed with exit code '1'
NOTE: Tasks Summary: Attempted 3700 tasks of which 3685 didn't need to be rerun and 1 failed.

Summary: 1 task failed:
  /srv/repos/projekte/smartwatch/asteroid/src/meta-swift-hybris/recipes-qt/qt5/qt5-qpa-hwcomposer-plugin_git.bb:do_compile
Summary: There were 2 ERROR messages shown, returning a non-zero exit code.
>>>

This was contained in the afforementioned logfile:
>>>
DEBUG: Executing shell function do_compile
NOTE: make -j 8 MAKEFLAGS=-j 8 OE_QMAKE_COMPILER=arm-oe-linux-gnueabi-gcc  -march=armv7ve -marm -mfpu=neon -mfloat-abi=hard --sysroot=/srv/repos/projekte/smartwatch/asteroid/build/tmp-glibc/work/swift-oe-linux-gnueabi/qt5-qpa-hwcomposer-plugin/5.6.1+gitrAUTOINC+a85d351855-r0/recipe-sysroot OE_QMAKE_CC=arm-oe-linux-gnueabi-gcc  -march=armv7ve -marm -mfpu=neon -mfloat-abi=hard --sysroot=/srv/repos/projekte/smartwatch/asteroid/build/tmp-glibc/work/swift-oe-linux-gnueabi/qt5-qpa-hwcomposer-plugin/5.6.1+gitrAUTOINC+a85d351855-r0/recipe-sysroot OE_QMAKE_CXX=arm-oe-linux-gnueabi-g++  -march=armv7ve -marm -mfpu=neon -mfloat-abi=hard --sysroot=/srv/repos/projekte/smartwatch/asteroid/build/tmp-glibc/work/swift-oe-linux-gnueabi/qt5-qpa-hwcomposer-plugin/5.6.1+gitrAUTOINC+a85d351855-r0/recipe-sysroot OE_QMAKE_CFLAGS= -O2 -pipe -g -feliminate-unused-debug-types -fdebug-prefix-map=/srv/repos/projekte/smartwatch/asteroid/build/tmp-glibc/work/swift-oe-linux-gnueabi/qt5-qpa-hwcomposer-plugin/5.6.1+gitrAUTOINC+a85d351855-r0=/usr/src/debug/qt5-qpa-hwcomposer-plugin/5.6.1+gitrAUTOINC+a85d351855-r0 -fdebug-prefix-map=/srv/repos/projekte/smartwatch/asteroid/build/tmp-glibc/work/swift-oe-linux-gnueabi/qt5-qpa-hwcomposer-plugin/5.6.1+gitrAUTOINC+a85d351855-r0/recipe-sysroot= -fdebug-prefix-map=/srv/repos/projekte/smartwatch/asteroid/build/tmp-glibc/work/swift-oe-linux-gnueabi/qt5-qpa-hwcomposer-plugin/5.6.1+gitrAUTOINC+a85d351855-r0/recipe-sysroot-native=  OE_QMAKE_CXXFLAGS= -O2 -pipe -g -feliminate-unused-debug-types -fdebug-prefix-map=/srv/repos/projekte/smartwatch/asteroid/build/tmp-glibc/work/swift-oe-linux-gnueabi/qt5-qpa-hwcomposer-plugin/5.6.1+gitrAUTOINC+a85d351855-r0=/usr/src/debug/qt5-qpa-hwcomposer-plugin/5.6.1+gitrAUTOINC+a85d351855-r0 -fdebug-prefix-map=/srv/repos/projekte/smartwatch/asteroid/build/tmp-glibc/work/swift-oe-linux-gnueabi/qt5-qpa-hwcomposer-plugin/5.6.1+gitrAUTOINC+a85d351855-r0/recipe-sysroot= -fdebug-prefix-map=/srv/repos/projekte/smartwatch/asteroid/build/tmp-glibc/work/swift-oe-linux-gnueabi/qt5-qpa-hwcomposer-plugin/5.6.1+gitrAUTOINC+a85d351855-r0/recipe-sysroot-native=  -fvisibility-inlines-hidden OE_QMAKE_LINK=arm-oe-linux-gnueabi-g++  -march=armv7ve -marm -mfpu=neon -mfloat-abi=hard --sysroot=/srv/repos/projekte/smartwatch/asteroid/build/tmp-glibc/work/swift-oe-linux-gnueabi/qt5-qpa-hwcomposer-plugin/5.6.1+gitrAUTOINC+a85d351855-r0/recipe-sysroot OE_QMAKE_LDFLAGS=-Wl,-O1 -Wl,--hash-style=gnu -Wl,--as-needed OE_QMAKE_AR=arm-oe-linux-gnueabi-ar OE_QMAKE_STRIP=echo OE_QMAKE_INCDIR_QT=/srv/repos/projekte/smartwatch/asteroid/build/tmp-glibc/work/swift-oe-linux-gnueabi/qt5-qpa-hwcomp [... cut ...]
Makefile:475: recipe for target '.obj/hwcomposer_backend_v11.o' failed
make: *** [.obj/hwcomposer_backend_v11.o] Error 1
make: *** Waiting for unfinished jobs....
ERROR: oe_runmake failed
WARNING: exit code 1 from a shell command.
ERROR: Function failed: do_compile (log file is located at /srv/repos/projekte/smartwatch/asteroid/build/tmp-glibc/work/swift-oe-linux-gnueabi/qt5-qpa-hwcomposer-plugin/5.6.1+gitrAUTOINC+a85d351855-r0/temp/log.do_compile.12708)
>>>

Please advise on how to proceed. :)

Thanks!

FlorentRevest, 2019-01-07:
Hey @Trooner,

Thanks a lot for your interest and let me apologize for my late reply as I was quite busy with other things.

The problem is that the current development version of our "prepare-build.sh" script fetches the latest commit of the master branch of meta-qt5. The master branch used to provide Qt 5.11 but it recently got updated to 5.12.
We also ship a recipe for qt5-qpa-hwcomposer-plugin whose code hasn't been updated to work with Qt 5.12 so there is an incompatibility between our qt5-qpa-hwcomposer-plugin and the version of Qt that is now fetched by meta-qt5 master.

There are two ways to approach this:
1- Trying to update the qt5-qpa-hwcomposer-plugin codebase to work with Qt 5.12. This is usually not trivial, especially if you don't have much experience with C++, Qt and OpenEmbedded. This is the kind of problems I take care of when I bump important layers (oe-core or meta-qt5) Note that there might be other stuff broken with Qt5.12 as well.
2- Reverting meta-qt5 back to a commit that fetches Qt 5.11. This is simple and straightforward.
You only need to add a commit hash at the end of this line (after "master") https://github.com/AsteroidOS/asteroid/blob/master/prepare-build.sh#L72 then remove your current src/meta-qt5 and source prepare-build.sh again. You should find a good commit hash in here https://github.com/meta-qt5/meta-qt5/commits/master Pick one before Qt got upgraded to 5.12. I would suggest 201fcf27cf07d60b7d6ab89c7dcefe2190217745 or 9fc65c05f3296b606d4ceb6f4670f5f95eecc379.
I don't have the time to build it myself right now so feel free to try it out and send a PR to https://github.com/AsteroidOS/asteroid with the meta-qt5 commit hash added to the previously mentioned line.
I'm really sorry for the inconvenience, this (pointing to the latest commit of master) is a bad design decision from my side that I accepted to take given that we are largely in a development cycle right now and I like breaking stuff.
Anyway, the guys from meta-webos-ports are not responsible for any of this. They would be happy to have a patch for qt5-qpa-hwcomposer-plugin as a side-effect of this issue but they are not to be blamed for the AsteroidOS build break... :)

Trooner, 2019-01-07:
Not a problem, I will need to wrap my head about openembedded structure anyhow if I'm going to work on the watch. I love that AsteroidOS exists, and although not perfect, its much better than android wear already. ;) 

So thanks for your great work up until now. We are all busy from time to time, so don't sweat it. :)

I'm a security engineer by trade, therefore I have done some C++ work. But it all but my primary language. ;) 

I will try to build a patch for qt5-qpa-hwcomposer-plugin and if I fail at that, I will revert the branch to a pre 1.12 commit. 

Thanks!

FlorentRevest, 2019-01-26:
The error has been temporarily fixed with https://github.com/AsteroidOS/asteroid/commit/70f5b2fb2d91f6cbaa6b058bf306c1417afc1894 (as long as we use Qt 5.11)
Let me know if you meet any problem with fixing the bluetooth on swift

closed 2019-01-26 by FlorentRevest