AsteroidOS/asteroid #82: Ambient display on wrist motion

Shadow53, 2019-01-13:
My Moto 360 watch with Android Wear supported peeking an ambient display for a few seconds if it detected the motion of me lifting my wrist to look at the screen. I've since switched to an LG G Watch running AsteroidOS and the lack of this feature is the biggest pain point, since I have to gesture on the screen a bit before it turns on. If there's something in my other hand, there's a hassle of setting down the thing or awkwardly rubbing the screen with my nose, just to look at the time.

I figure a combination of input from the accelerometer and/or the gyroscope would give the necessary information to determine the user is looking at their watch. I can collect some example data from own watch, if that will help any; I would just need some direction on how to do that.

njfox, 2019-12-03:
> Every WearOS watch comes with a device that can wake up the watch when a movement is detected. On dory, it is the lis3dsh.

Do you know offhand what device it is in bass?

FlorentRevest, 2019-12-04:
No, but maybe you can find that using test_sensors

jrtberlin, 2020-09-18:
We have support for [tilt-to-wake](https://github.com/AsteroidOS/mce/pull/6) and [allways-on](https://github.com/AsteroidOS/meta-asteroid/pull/43) now.

closed 2020-09-18 by jrtberlin