AsteroidOS/asteroid #83: [dory] [nightly] Bluetooth doesn't turn on

jrtberlin, 2019-02-14:
1. Bluetooth is not possible to enable on every boot.
2. When Bluetooth is shown as enabled, it can not be discovered by any other device with and without AsteroidOSSync.
3. I have not experienced this issue with the 1.0 release.

Here are a few logs:

```
ceres@dory:~$ dmesg | grep bluetooth
[    0.577315] rfkill: bluetooth_rfkill_probe: entry
[    0.577408] rfkill: bluetooth_set_power: set blocked 1
[    0.577438] rfkill: bluetooth_set_power: Bluetooth OFF!!
[    3.041770] rfkill: bluetooth_set_power: set blocked 0
[    3.098059] rfkill: bluetooth_set_power: Bluetooth ON!!
[    7.840918] rfkill: bluetooth_set_power: set blocked 0
[    7.894676] rfkill: bluetooth_set_power: Bluetooth ON!!

```

```
ceres@dory:~$ bluetoothctl
Agent registered
[bluetooth]# discoverable on
Failed to set discoverable on: org.bluez.Error.Failed

```

```
root@dory:ceres# journalctl | grep bluetooth
Feb 09 18:47:58 dory kernel: rfkill: bluetooth_rfkill_probe: entry
Feb 09 18:47:58 dory kernel: rfkill: bluetooth_set_power: set blocked 1
Feb 09 18:47:58 dory kernel: rfkill: bluetooth_set_power: Bluetooth OFF!!
Feb 09 18:47:59 dory systemd[1]: Starting Load firmware into BCM20715A1 bluetooth chip...
Feb 09 18:47:59 dory kernel: rfkill: bluetooth_set_power: set blocked 0
Feb 09 18:47:59 dory kernel: rfkill: bluetooth_set_power: Bluetooth ON!!
Feb 09 18:47:59 dory systemd[1]: Started Load firmware into BCM20715A1 bluetooth chip.
Feb 09 18:48:04 dory kernel: rfkill: bluetooth_set_power: set blocked 0
Feb 09 18:48:04 dory kernel: rfkill: bluetooth_set_power: Bluetooth ON!!
Feb 09 18:48:05 dory bluetoothd[388]: Bluetooth daemon 5.50
Feb 09 18:48:05 dory bluetoothd[388]: Bluetooth management interface 1.9 initialized
Feb 09 19:00:28 dory bluetoothd[388]: Failed to set mode: Failed (0x03)
Feb 09 19:00:33 dory bluetoothd[388]: Failed to set mode: Failed (0x03)
Feb 09 19:01:04 dory bluetoothd[388]: Failed to set mode: Failed (0x03)
Feb 09 19:01:18 dory bluetoothd[388]: Failed to set mode: Failed (0x03)

```

```
root@dory:ceres# journalctl | grep btsyncd
Feb 09 18:48:05 dory dbus-daemon[297]: [system] Activating via systemd: service name='org.bluez' unit='dbus-org.bluez.service' requested by ':1.25' (uid=1000 pid=365 comm="/usr/bin/asteroid-btsyncd ")
Feb 09 18:48:05 dory asteroid-btsyncd[365]: Service "org.bluez" is running
Feb 09 18:48:05 dory asteroid-btsyncd[365]: No BLE adapter found
Feb 09 18:48:05 dory asteroid-btsyncd[365]: QDBusConnection: name 'org.bluez' had owner '' but we thought it was ':1.29'
Feb 09 18:48:05 dory asteroid-btsyncd[365]: Service "org.bluez" is running
Feb 09 18:48:09 dory asteroid-btsyncd[365]: BLE Adapter "/org/bluez/hci0" found
Feb 09 18:48:09 dory asteroid-btsyncd[365]: Service "/" registered
Feb 09 18:48:09 dory asteroid-btsyncd[365]: Advertisement "/org/asteroidos/btsyncd/advertisement" registered
```

jrtberlin, 2019-02-22:
Log for /usr/bin/brcm_patchram_plus -d ...

```
ceres@dory:~$ su
root@dory:ceres# /usr/sbin/rfkill unblock bluetooth
root@dory:ceres# /usr/bin/brcm_patchram_plus -d --enable_lpm --enable_hci --no2bytes --patchram /vendor/firmware/BCM20715A1.hcd /dev/ttyHS99
option enable_lpm
option enable_hci
option no2bytes
option patchram with arg /vendor/firmware/BCM20715A1.hcd
/dev/ttyHS99 
writing
01 03 0c 00 
writing
01 03 0c 00 
writing
01 03 0c 00 
writing
01 03 0c 00 
writing
01 03 0c 00 
writing
01 03 0c 00 
writing
01 03 0c 00 
writing
01 03 0c 00 
writing
01 03 0c 00 
writing
01 03 0c 00 
writing
01 03 0c 00 
writing
01 03 0c 00 
writing
01 03 0c 00 
writing
01 03 0c 00 
writing
01 03 0c 00 
writing
01 03 0c 00 
writing

```

jrtberlin, 2019-02-25:
Maybe related issue: https://bbs.csdn.net/topics/390719686?list=lz (Chinese)

gllmhyt, 2019-03-07:
I am experiencing the same issue with the same outputs. I can provide logs and do some testings too if it is needed to solve this unfortunate problem.

jrtberlin, 2019-04-29:
**Some additional logging**
Syscalls of bluetoothctl discoverable on (after agent is registered):

```
read(0, "\r", 1)                        = 1
write(1, "\n", 1
)                       = 1
ioctl(0, SNDCTL_TMR_STOP or TCSETSW, {B38400 opost isig icanon echo ...}) = 0
rt_sigaction(SIGINT, {SIG_DFL, [], SA_RESTORER, 0xb6ce6481}, {0xb6dd13b5, [], SA_RESTORER, 0xb6ce6481}, 8) = 0
rt_sigaction(SIGTERM, {SIG_DFL, [], SA_RESTORER, 0xb6ce6481}, {0xb6dd13b5, [], SA_RESTORER, 0xb6ce6481}, 8) = 0
rt_sigaction(SIGHUP, {SIG_DFL, [], SA_RESTORER, 0xb6ce6481}, {0xb6dd13b5, [], SA_RESTORER, 0xb6ce6481}, 8) = 0
rt_sigaction(SIGQUIT, {SIG_DFL, [], SA_RESTORER, 0xb6ce6481}, {0xb6dd13b5, [], SA_RESTORER, 0xb6ce6481}, 8) = 0
rt_sigaction(SIGALRM, {SIG_DFL, [], SA_RESTORER, 0xb6ce6481}, {0xb6dd13b5, [], SA_RESTORER, 0xb6ce6481}, 8) = 0
rt_sigaction(SIGTSTP, {SIG_DFL, [], SA_RESTORER, 0xb6ce6481}, {0xb6dd13b5, [], SA_RESTORER, 0xb6ce6481}, 8) = 0
rt_sigaction(SIGTTOU, {SIG_DFL, [], SA_RESTORER, 0xb6ce6481}, {0xb6dd13b5, [], SA_RESTORER, 0xb6ce6481}, 8) = 0
rt_sigaction(SIGTTIN, {SIG_DFL, [], SA_RESTORER, 0xb6ce6481}, {0xb6dd13b5, [], SA_RESTORER, 0xb6ce6481}, 8) = 0
rt_sigaction(SIGWINCH, {SIG_DFL, [], SA_RESTORER, 0xb6ce6481}, {0xb6dd0d3d, [], SA_RESTORER|SA_RESTART, 0xb6ce6481}, 8) = 0
gettimeofday({1556551332, 800149}, NULL) = 0
write(1, "No default controller available\n", 32No default controller available
) = 32
ioctl(0, TIOCGWINSZ, {ws_row=67, ws_col=314, ws_xpixel=0, ws_ypixel=0}) = 0
ioctl(0, TIOCSWINSZ, {ws_row=67, ws_col=314, ws_xpixel=0, ws_ypixel=0}) = 0
ioctl(0, TCGETS, {B38400 opost isig icanon echo ...}) = 0
ioctl(0, SNDCTL_TMR_STOP or TCSETSW, {B38400 opost isig -icanon -echo ...}) = 0
write(1, "\33[0;94m[bluetooth]\33[0m# ", 24[bluetooth]# ) = 24
```

fizzgig242, 2019-06-18:
Same issue here. I am not able to connect to the phone.

mauricioduarte01, 2020-01-23:
Hello. I am experiencing the same issue on my ZenWatch 2 (sparrow). Any plans on fixing/updating this project?
Thank you for the effort

harshitthakore, 2020-02-24:
Hello, got the same issue on Asus zen watch 3.

jrtberlin, 2020-02-24:
@harshitthakore @mauricioduarte01 Can you reproduce [this](https://github.com/AsteroidOS/asteroid/issues/83#issuecomment-466514115) with the same results? And `journalctl | grep bluetooth`? I think that your issues are unrelated to this issue, as I suspect this one to be hardware-specific.

mauricioduarte01, 2020-02-24:
@jrtberlin Hi, I think my issue is directly related to: https://github.com/AsteroidOS/meta-sparrow-hybris/issues/4 . I will try to debug later today and post the log. thank you

harshitthakore, 2020-02-24:
@jrtberlin I am not sure I am on right thread or not. My watch successfully boots up and when I try to enable the Bluetooth in settings its shows off but at the notification, it's enabled.

jrtberlin, 2020-02-24:
@harshitthakore So the symptoms you are experiencing are identical. Can you check if you get the same logs? Which version of AsteroidOS are you running (stable 1.0 or nightly + build date) and is the issue gone after a reboot?
I think the underlying issue is not the same. (I could be wrong tho, that's why I'm asking)

harshitthakore, 2020-02-24:
@jrtberlin I will work on log later today.
Os is 1.1 nightly and I am having the same issue after a reboot also.

harshitthakore, 2020-02-25:
@jrtberlin Here is the log I got after executing "journalctl | grep bluetooth" .
root@swift:ceres# journalctl | grep bluetooth
Feb 12 16:54:16 swift bluetoothd[526]: Bluetooth daemon 5.50
Feb 12 16:54:16 swift bluetoothd[526]: Failed to open RFKILL control device
Feb 12 16:54:16 swift bluetoothd[526]: Bluetooth management interface 1.7 initialized

MagneFire, 2020-02-25:
@harshitthakore Your problem is not related to this issue.

The Asus Zenwatch 3 does not support Bluetooth yet(https://asteroidos.org/install/swift/ see the support table).

This is the issue actually related to your watch: https://github.com/AsteroidOS/meta-swift-hybris/issues/1

jrtberlin, 2020-02-25:
Thanks @MagneFire I will hide the messages by @harshitthakore to keep the history clean with only related issues.

jrtberlin, 2020-05-27:
It seems the issue has been introduced with a systemd update. We [removed systemd-rflill](https://github.com/AsteroidOS/meta-asteroid/pull/39) and I can not reproduce the issue on the latest nightly. Everyone who had issues on dory is encouraged to retest with the [latest nightly](https://release.asteroidos.org/nightlies/dory/).

closed 2020-05-27 by FlorentRevest

gllmhyt, 2020-06-03:
Even with the latest nightly update, it seems that Bluetooth cannot be turned on still.