AsteroidOS/asteroid #85: [Question] Activate vibration via command line

pleft, 2019-03-07:
Is there any way to enable vibration via command line (harmony build)? I accidentally dropped my watch and it has a big crack across the screen and it seems that the screen region where the vibration button is in the quick settings is no longer responsive. And to my bad luck I had just disabled vibration prior dropping it.

FlorentRevest, 2019-03-07:
Could you try to run as ceres the following command ?
dbus-send --type=method_call --dest=com.nokia.profiled /com/nokia/profiled com.nokia.profiled.set_profile string:"general"

pleft, 2019-03-08:
Hi Florent, tried:

```
ceres@harmony:~$ dbus-send --type=method_call --dest=com.nokia.profiled /com/nokia/profiled com.nokia.profiled.set_profile string:"general"
Failed to open connection to "session" message bus: Failed to connect to socket /dbus/user_bus_socket: No such file or directory
ceres@harmony:~$
```

FlorentRevest, 2019-03-10:
Run `export DBUS_SESSION_BUS_ADDRESS=unix:path=/run/user/1000/bus` first.
This should be fixed by https://github.com/AsteroidOS/meta-asteroid/commit/78d382f20f2e956e5de5283f3d8cd61a5145e941 if you ssh directly to the ceres user. (ie: not to root and then su to ceres)

pleft, 2019-03-11:
Great it worked!!! Many thanks, at least my watch is functional again (with only the half touchscreen working :))

Another request, it there any way to enable usb abd mode via ssh? cause the "ok" button is in the non-touchable area...

pleft, 2019-03-11:
found a workaround. Leaving it here in case anyone finds it usefull.

ssh as root, create a script file e.g. `adb_start.sh`.

```
nohup /usr/bin/android-gadget-setup adb; nohup/usr/bin/adbd
```
then execute the script

```
./adb_start.sh
```

ssh connection will immediatelly drop and adb service will start. Now on the host computer
```
adb devices
```
should show the id of the asteroidos watch.

closing the issue, thanks for the help!

closed 2019-03-11 by pleft

FlorentRevest, 2019-03-12:
Regarding adb, I would suggest using usb-moded-util instead https://github.com/philippedeswert/usb-moded/blob/master/src/usb_moded-util.c#L399