AsteroidOS/asteroid #86: Question on compatibility with Android (non-Wear) smartwatches

eimiveritas, 2019-03-12:
Hi, I would like to ask if there is any existing solution to install AsteroidOS on android (non-wear) smartwatches? I have read https://asteroidos.org/wiki/porting-status/ where it states that "It might be more difficult to get access to the kernel’s or android tree of those devices but libhybris could also make running AsteroidOS on top of them possible.". I am looking at zeblaze thor 4 pro specifically, as I own one of those and want to install AsteroidOS on it.

FlorentRevest, 2019-03-12:
Hi, as stated in the porting status page, porting AsteroidOS to a watch requires at least access to its kernel's source code. The Zeblaze Thor 4 Pro uses a MTK6739 SoC. Mediatek does not publish the source code of their kernels.
We got our hands on a kernel for MTK6580 watches thanks to the @OpenWatchProject but as far as I know it is not compatible with the MTK6739 and your Zeblaze Thor 4 Pro.
Maybe @deadman96385 can comment on this?

deadman96385, 2019-03-12:
I only have access to the mt6580 kernel source and the mt6737 kernel source. mt6739 is still really new for the smartwatch world so it will probably be awhile before I can get it sadly.

pankajbca, 2020-05-19:
> I only have access to the mt6580 kernel source and the mt6737 kernel source. mt6739 is still really new for the smartwatch world so it will probably be awhile before I can get it sadly.

@deadman96385 Hi, Can you please provide me the Kernal source of mt6737. I have mt6737 watch and wanna port astroid os on it.

pankajbca, 2021-03-02:
@deadman96385 Hi, Can you please provide me the Kernal source of mt6737. I have an mt6737 watch and wanna port astroid os on it. I have an MT6737 android watch and wanted to install asteroidOS on it. please provide me the kernal sources of the MT6737 chip.