AsteroidOS/asteroid #88: FAILED (remote: Unknown chunk type) on mooneye

bellicapax, 2019-05-15:
This seems to be related to [https://github.com/AsteroidOS/asteroid/issues/31](url). Built asteroid os for mooneye, tried to flash it via fastboot and got the following error:

` fastboot flash userdata asteroid-image-mooneye.ext4
target reported max download size of 134217728 bytes
Invalid sparse file format at header magic
erasing 'userdata'...
OKAY [  1.258s]
sending sparse 'userdata' 1/4 (131068 KB)...
OKAY [ 31.393s]
writing 'userdata' 1/4...
FAILED (remote: Unknown chunk type)
finished. total time: 32.701s
`

FlorentRevest, 2019-05-15:
Err, I would really not recommend flashing mooneye with the "real install" method... (This is why it is not documented on the website yet)
Anyway, you'd probably need a backup of a Adnroid boot partition (I probably have one somewhere if you can't find one on XDA but I don't know if this will be the right version for you)
Also, watches usually have a way to enter the bootloader when the kernel doesn't boot (otherwise devs could not work). You would need that to flash the boot partition backup. Usually, it is either a weird usage of the power button or a weird usage of the screen. You'd probably need to browse the XDA-Dev forum for mooneye and see if you find something.

bellicapax, 2019-05-24:
Thanks for the fast reply! I'm looking into using SPFlashtool but I can't find a scatter file anywhere and have been unable to find any directions for how to enter the bootloader manually.

bellicapax, 2019-05-24:
Update: I found a scatter file that allowed me to format my watch and then loaded WearOS onto my watch via SP Flash Tool [here](https://forum.xda-developers.com/smartwatch/other-smartwatches/ticwatch-e-chinese-version-ticwear-t3781183) so I could get back to the bootloader.

I performed the steps in the other thread to create my ext4.sparse file and then performed the flash command with the following error:

```
fastboot flash userdata asteroid-image-mooneye.ext4.sparse 
target reported max download size of 134217728 bytes
erasing 'userdata'...
FAILED (remote: failed to get format permission for partition 'userdata'
)
```
I was also unable to `fastboot erase userdata` or `fastboot format userdata`. I rebooted into WearOS and pushed the file for a temporary boot. That managed to work and I was in AsteroidOS for a brief moment though swiping up and down was extremely tricky. When the watch slept it turned off (not sure if that's intentional?)

FlorentRevest, 2019-05-27:
Mhh this is interesting...
Could you, however, try to flash the AsteroidOS userdata partition (not sparse) from SPFlashTool ? (as long as you know how to revert it back later on)

bellicapax, 2019-06-25:
I managed to get AsteroidOS on the watch by performing these steps:

1. Load the scatter file for Android WearOS
2. Replace the userdata with AsteroiOS & deselect the boot img
3. Format except bootloader (bc I had already flashed the bootloader)
4. Download the selected files

After that, it was able to boot into AsteroidOS, though these major problems still remained:

1. Can't connect via Bluetooth
2. OS will randomly shut itself down
3. Swiping is very difficult

FlorentRevest, 2019-07-20:
Thanks for the guide @bellicapax .
Could you please provide the scatter file that you mentioned in your short guide?

If you try the next autobuild: (or build it yourself)
I improved the Bluetooth support. We are now able to discover the watch from a phone and establish a BT connection but there are still problems to actually use AsteroidOSSync.
I also fixed most touchscreen issues, including the one where touchscreen would just not work for the first 30secs when the watch is booted. I also find it much easier to swipe now.
I also used to experience reboots, typically when disconnecting the watch from USB. I have not been able to reproduce those bugs yet.
I also fixed the vibrator support.

jonshipman, 2019-09-10:
Happened onto this thread googling as I'm currently fighting a mooneye that I tried Ticwear on and trying to restore a TWRP backup.

From my findings, it seems that stock, otb, mooneye has flags on the partitions:
Specifically:

    Partition Logs:
    /boot | /dev/block/mmcblk0p7 | Size: 14MB
    Flags: Can_Be_Backed_Up IsPresent Can_Flash_Img 
    Primary_Block_Device: /dev/block/mmcblk0p7
    Display_Name: Boot
    Storage_Name: boot
    Backup_Path: /boot
    Backup_Name: boot
    Backup_Display_Name: Boot
    Storage_Path: /boot
    Current_File_System: emmc
    Fstab_File_System: emmc
    Backup_Method: dd

    /recovery | /dev/block/mmcblk0p8 | Size: 32MB
    Flags: IsPresent Can_Flash_Img 
    Primary_Block_Device: /dev/block/mmcblk0p8
    Display_Name: Recovery
    Storage_Name: recovery
    Backup_Path: /recovery
    Backup_Name: recovery
    Backup_Display_Name: Recovery
    Storage_Path: /recovery
    Current_File_System: emmc
    Fstab_File_System: emmc
    Backup_Method: dd

    /cache | /dev/block/mmcblk0p15 | Size: 58MB Used: 0MB Free: 58MB Backup Size: 0MB
    Flags: Can_Be_Mounted Can_Be_Wiped Can_Be_Backed_Up Wipe_During_Factory_Reset Wipe_Available_in_GUI IsPresent 
    Primary_Block_Device: /dev/block/mmcblk0p15
    Display_Name: Cache
    Storage_Name: Cache
    Backup_Path: /cache
    Backup_Name: cache
    Backup_Display_Name: Cache
    Storage_Path: /cache
    Current_File_System: ext4
    Fstab_File_System: ext4
    Backup_Method: files

    /system | /dev/block/mmcblk0p14 | Size: 1007MB Used: 690MB Free: 317MB Backup Size: 690MB
    Flags: Can_Be_Mounted Can_Be_Wiped Can_Be_Backed_Up Wipe_Available_in_GUI IsPresent 
    Primary_Block_Device: /dev/block/mmcblk0p14
    Display_Name: System
    Storage_Name: System
    Backup_Path: /system
    Backup_Name: system
    Backup_Display_Name: System
    Storage_Path: /system
    Current_File_System: ext4
    Fstab_File_System: ext4
    Backup_Method: files

    /data | /dev/block/mmcblk0p17 | Size: 2404MB Used: 784MB Free: 1584MB Backup Size: 784MB
    Flags: Can_Be_Mounted Can_Be_Wiped Can_Be_Backed_Up Wipe_During_Factory_Reset Wipe_Available_in_GUI IsPresent Can_Be_Encrypted Has_Data_Media Can_Encrypt_Backup Use_Userdata_Encryption Is_Storage Is_Settings_Storage 
    Symlink_Path: /data/media/0
    Symlink_Mount_Point: /sdcard
    Primary_Block_Device: /dev/block/mmcblk0p17
    Length: -16384
    Display_Name: Data
    Storage_Name: Internal Storage
    Backup_Path: /data
    Backup_Name: data
    Backup_Display_Name: Data
    Storage_Path: /data/media/0
    Current_File_System: ext4
    Fstab_File_System: ext4
    Backup_Method: files
    MTP_Storage_ID: 65537

I am unsure if this is the "real" reason, but it would make sense if these flags aren't present you'd get a permissions error.

closed 2026-03-14 by moWerk