AsteroidOS/asteroid #89: pulseaudio-modules build error causing asteroid-image build to fail

derkommutator, 2019-05-22:
I'm trying to build asteroid-image for the mooneye platform and bitbake is failing to build pulseaudio-modules, quitting with the following error:

ERROR: pulseaudio-modules-droid-+gitAUTOINC+e28d1e22d3-r1 do_compile: oe_runmake failed
[...]
| In file included from droid-util.c:60:
| droid-util.h:49:2: error: #error "No valid ANDROID_VERSION found."
|  #error "No valid ANDROID_VERSION found."

I've also tried building for the bass platform (which I've built successfully in the past), and get the same error.  Furthermore, I have tried a totally fresh clone on a different build box, with the same results.

Has something perhaps changed in the underlying pulseaudio-modules-droid repository that is now causing it to fail to build in the AsteroidOS build environment?

Full error dump attached, although almost everything in it is just errors caused by the original error causing the build not to include a required header file, so not really relevant to the actual problem here.

[log.do_compile.29730.txt](https://github.com/AsteroidOS/asteroid/files/3206106/log.do_compile.29730.txt)

FlorentRevest, 2019-05-27:
Thank you for reporting @derkommutator !
This should now be fixed by https://github.com/AsteroidOS/meta-asteroid/commit/14f2072f4bcedc70efbfa1f12b2384ba0bbc02b6

Feel free to hop in to IRC if you want to help with the mooneye port! It is still a little bit rough on the edges at this point and we are strongly interested in contributions.

derkommutator, 2019-05-28:
I confirm that https://github.com/AsteroidOS/meta-asteroid/commit/14f2072f4bcedc70efbfa1f12b2384ba0bbc02b6 fixes the pulseaudio problem.  Now the build fails on the do_install phase of mlite, again on both build boxes.

Log attached, but I'll close this issue, since the mlite problem isn't technically the same issue.  Something to look at though.

[log.do_install.7734.txt](https://github.com/AsteroidOS/asteroid/files/3226017/log.do_install.7734.txt)

closed 2019-05-28 by derkommutator

FlorentRevest, 2019-05-28:
For the mlite build issue please apply this patch locally to your meta-qt5 https://github.com/meta-qt5/meta-qt5/pull/206
I am just waiting for it to be merged upstream as there is no clean way to solve this downstream

derkommutator, 2019-05-28:
Aha, thanks for hunting that down.  I figured it was that simple.  That does indeed fix the problem.