AsteroidOS/asteroid #90: Time and Date not syncing in dory

heavyinfo, 2019-06-15:
Hi,

Time and Date not syncing in Dory after installing **12th May nightly**. The time starts from the default & progresses with it. Date stays at 3rd June, even if the time progresses.

Tried disabling and enabling 'Synchronize time' from Android sync app multiple times and reinstalled sync app with no effect.

heavyinfo, 2019-06-18:
Time sync seems to be fixed in the **17th June nightly**.

FlorentRevest, 2019-07-02:
Thanks for reporting... Unfortunately nothing has been done to fix this bug so I am not sure what happened :)

jrtberlin, 2019-07-02:
The nightly from 24.6 worked great for a few days. Yesterday date and time broke again after a reboot (although nothing changed except the actual time and date) and even a reflash of the same nightly and the one of 1st July did not fix the issue. I would keep this issue open.

heavyinfo, 2019-07-11:
My 17th June nightly has survived several restarts, time & date sync hasn't been an issue so far.

mrbijan, 2019-07-20:
Same issue with Harmony (KW88).

closed 2026-03-14 by moWerk