AsteroidOS/asteroid #91: [bass] Black watch face after navigation

StefWe, 2019-07-17:
-    Device: bass
-    AsteroidOS: Version 1.0

**Steps to reproduce**
I'm sorry, but I haven't recognized a pattern yet to repeat it.
I'm in the settings or somewhere else in the menu and when I come back to the watch face it's completely black.
If i go to the settings and switch the watch face to another and back, it's fixed.

**Expected behavior**
The normal watch face what i have selected.

**Actual behavior**
Watch face is completely black. Only the date is readable.

derkommutator, 2019-07-17:
Can confirm that I have seen this once as well, on bass.  No particular course of action seemed to have prompted it, and it's never repeated.

moWerk, 2026-03-14:
Not reported in years, closing.