AsteroidOS/asteroid #92: [Lenok][Nightly] UI lagging in apps while finger on touchscreen

alphazero1990, 2019-08-10:
- Device: LG G Watch R
- AsteroidOS: Snapshot-20190805

When in one of the apps (calendar or settings for instance), dragging around makes the UI stutter (framerate is lower). As soon as I lift my finger from the touchscreen the framerate comes back to smooth 60 fps. Watchface and main menu is unaffected.

TLDR: UI stutters when finger on touchscreen inside apps.

FlorentRevest, 2019-08-25:
Can you maybe try to turn one more CPU core on ?
For instance, remove this line https://github.com/AsteroidOS/meta-lenok-hybris/blob/master/recipes-core/underclock/underclock/underclock#L6 in /usr/bin/underclock
You can also try to change the CPU governor to something more agressive, but in any case your battery life should be shorter. https://github.com/AsteroidOS/meta-lenok-hybris/blob/master/recipes-core/underclock/underclock/underclock#L4

beckerichmika, 2019-09-14:
Is it because of the clock rate or because of the touchscreen refresh rate? An incorrectly calibrated screen or a driver which is polling the touch state too often might drive the ui nuts

eLtMosen, 2025-04-24:
Tested and found to be fixed

closed 2025-04-24 by eLtMosen