AsteroidOS/asteroid #93: [Harmony] Screen unresponsive

GoDieNow, 2019-08-11:
- Device: (Harmony) Zeblaze Thor
- AsteroidOS: nightly 05-08-2019

Right after the install the screen is completely unresponsive, I'm not even able to move beyond the language selection screen.

FlorentRevest, 2019-08-25:
I don't have a Zeblaze Thor to test this out so I just added a warning on the website. All the "harmony" watches are very similar but not exactly the same so they might come up with problems. https://asteroidos.org/install/harmony/

Amok83, 2020-01-31:
Hello I have the same problem with the "harmony" kw88 pro, screen works but with one finger touching the top and scrolling with the other one, can you please fix the problem. Thanks on advance