AsteroidOS/asteroid #94: [Harmony] [KW99] Battery indicator doesn't work

holly6k, 2019-09-07:
Device: KW99
AsteroidOS: nightly 20190902

Battery indicator shows "undefined %" and charging status is unknown (no notification after plugging in).

JNissi, 2019-09-10:
`cat /sys/class/power_supply/battery/uevent` (or `/capacity`) shows the battery status correctly. Will debug DBus next.

JNissi, 2019-09-26:
I flashed the latest nightly (23.9.2019), still the same problem.

I then flashed a bare-bones build where I've removed most stuff so it only starts the psplash and sshd.

After testing that image around for a moment I flashed the latest Asteroid nightly back on the watch and for some reason the battery level is now shown correctly. 

I wish I could give any kind of explanation for what happened.

FlorentRevest, 2019-09-27:
This very much looks like a similar class of problem as what we had here https://github.com/AsteroidOS/meta-bass-hybris/issues/6#issuecomment-525008899

If my intuition is right: the battery level is provided by statefs providers. statefs providers are registered in a post-inst script run on first boot by opkg https://github.com/AsteroidOS/meta-asteroid/blob/master/recipes-nemomobile/statefs/statefs_git.bb#L51 and for some weird reason, since a few releases, opkg just crashes randomly while trying to run "opkg reconfigure" on first boot. This really happens randomly and is virtually impossible to reproduce. Since opkg reconfigure is only ran once, if it fails then the commands it is supposed to run are not run again.

I have no explanation for why this happens and no good fix for it. I was able to mitigate the related time bug (a postinst script to setcap timed-qt5 was used) by moving the command to build-time https://github.com/AsteroidOS/meta-asteroid/commit/980632104da0c139150ccd50c54a28aabae8d10a But afaik this is not something we can do with statefs providers installation.

MrNickH, 2020-02-06:
> This very much looks like a similar class of problem as what we had here [AsteroidOS/meta-bass-hybris#6 (comment)](https://github.com/AsteroidOS/meta-bass-hybris/issues/6#issuecomment-525008899)
> 
> If my intuition is right: the battery level is provided by statefs providers. statefs providers are registered in a post-inst script run on first boot by opkg https://github.com/AsteroidOS/meta-asteroid/blob/master/recipes-nemomobile/statefs/statefs_git.bb#L51 and for some weird reason, since a few releases, opkg just crashes randomly while trying to run "opkg reconfigure" on first boot. This really happens randomly and is virtually impossible to reproduce. Since opkg reconfigure is only ran once, if it fails then the commands it is supposed to run are not run again.
> 
> I have no explanation for why this happens and no good fix for it. I was able to mitigate the related time bug (a postinst script to setcap timed-qt5 was used) by moving the command to build-time [AsteroidOS/meta-asteroid@9806321](https://github.com/AsteroidOS/meta-asteroid/commit/980632104da0c139150ccd50c54a28aabae8d10a) But afaik this is not something we can do with statefs providers installation.

Just ran
`  /usr/lib/statefs/provider-action register /usr/lib/statefs/libprovider-power_udev.so default --system`
And got 
```
Trying to dump default provider "/usr/lib/statefs/libprovider-power_udev.so"
Trying to dump as a loader
Not a loader: /usr/lib/statefs/libprovider-power_udev.so
File /usr/lib/statefs/libprovider-power_udev.so is not config?, skipping
Can't retrieve information from libprovider-power_udev.so
```

Not sure if helping or barking up the wrong tree here though.

schindlershadow, 2021-08-10:
I managed to get this on sturgeon 
[image]
im on 1.1-nightly

speaktrap, 2021-09-20:
I also encountered this issue, fresh flash after unlocking bootloader of a stock Watch.
I switched to ADB, did the _opkg reconfigure statefs_ and rebooted, works like a charm.
Why not implement some kind of quick hack for now? Something akin to "if battery status undefined –> try to reregister power status provider"

MagneFire, 2022-01-23:
Since https://github.com/AsteroidOS/meta-asteroid/pull/83 has been merged. So this issue is finally solved :tada:

closed 2022-01-23 by MagneFire