AsteroidOS/asteroid #96: LG Urbane in boot loop, possibly bricked

frankgerhardt, 2019-10-09:
I followed the instructions in https://asteroidos.org/install/bass/ using Debian 9. 

Flashing worked fine:I followed the instrucktions in https://asteroidos.org/install/bass/ using Debian 9. 

Flashing worked fine:

```
$ adb devices
List of devices attached
* daemon not running. starting it now on port 5037 *
* daemon started successfully *
504KPLC0044952	unauthorized

$ adb reboot bootloader

$ fastboot flash userdata asteroid-image-bass.ext4
target reported max download size of 268435456 bytes
Invalid sparse file format at header magi
erasing 'userdata'...
OKAY [  2.275s]
sending sparse 'userdata' 1/2 (262140 KB)...
OKAY [  8.682s]
writing 'userdata' 1/2...
OKAY [ 56.432s]
sending sparse 'userdata' 2/2 (174900 KB)...
OKAY [  5.792s]
writing 'userdata' 2/2...
OKAY [143.314s]
finished. total time: 216.494s

$ fastboot flash boot zImage-dtb-bass.fastboot
target reported max download size of 268435456 bytes
sending 'boot' (17754 KB)...
OKAY [  0.673s]
writing 'boot'...
OKAY [  1.374s]
finished. total time: 2.047s

$ fastboot continue
resuming boot...
OKAY [  0.002s]
finished. total time: 0.002s
```

After the last command, the watch booted, and rebooted, and rebooted. Adb can not find the watch any more. 

The watch is connected to the charger. I can tell if it is fully chared or not. It was in the boot loop for many hours. The watch is quite hot (overheating?). Is the battery icon acting like a progress bar? 

Can I somehow get into the fastboot menu again, to try flashing again? Any key "combination" for 

```
$ adb devices
List of devices attached
* daemon not running. starting it now on port 5037 *
* daemon started successfully *
504KPLC0044952	unauthorized

$ adb reboot bootloader

$ fastboot flash userdata asteroid-image-bass.ext4
target reported max download size of 268435456 bytes
Invalid sparse file format at header magi
erasing 'userdata'...
OKAY [  2.275s]
sending sparse 'userdata' 1/2 (262140 KB)...
OKAY [  8.682s]
writing 'userdata' 1/2...
OKAY [ 56.432s]
sending sparse 'userdata' 2/2 (174900 KB)...
OKAY [  5.792s]
writing 'userdata' 2/2...
OKAY [143.314s]
finished. total time: 216.494s

$ fastboot flash boot zImage-dtb-bass.fastboot
target reported max download size of 268435456 bytes
sending 'boot' (17754 KB)...
OKAY [  0.673s]
writing 'boot'...
OKAY [  1.374s]
finished. total time: 2.047s

$ fastboot continue
resuming boot...
OKAY [  0.002s]
finished. total time: 0.002s
```

After the last command, the watch booted, and rebooted, and rebooted. Adb can not find the watch. 

The watch is connected to the charger. I can tell if it is fully chared or not. It was in the boot loop for many hours. Is the battery icon acting like a progress bar? 

Can I somehow get into the fastboot menu again, to try flashing again? Any key "combination" for pressing the single button?

[image]

wgahnagl, 2019-10-09:
Swipe down from the top left corner to the bottom right to open up the boot menu 👌

frankgerhardt, 2019-10-09:
Thank you! That worked. I wiped everything and did a factory reset. Then I did not `fastboot continue` but `Start` from the bootloader.

closed 2019-10-09 by frankgerhardt